bdring/Grbl_Esp32 #1015: Translate Machine.h for TMPCNC

whins, 2021-12-27:
This issue will automatically convert a Grbl_Esp32 "machine.h" file to FluidNC YAML format.

Change YOUR NAME in the title to some string that identifies your machine configuration.

Attach your Machine.h file here - but first change the .h extension to .txt because GitHub will not allow you to attach a file with a .h extension.  Submit the issue, and after a few minutes, a new comment will appear with the FluidNC YAML file. 
[tmpcnc.txt](https://github.com/bdring/Grbl_Esp32/files/7777749/tmpcnc.txt)

MitchBradley, 2021-12-27:
Something went wrong with the converter script so I ran it manually to get the following output - Mitch Bradley

```
name: ESP32_V4
board: unknown

kinematics:
  Cartesian:

stepping:
  engine: RMT
  idle_ms: 250
  dir_delay_us: 0
  pulse_us: 3
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.13
  x:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 300.000
    soft_limits: false
    motor0:
      limit_all_pin: gpio.36:low
      hard_limits: false
      stepstick:
        direction_pin: gpio.12
        step_pin: gpio.14
    motor1:
      null_motor:
  y:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 300.000
    soft_limits: false
    motor0:
      limit_all_pin: gpio.39:low
      hard_limits: false
      stepstick:
        direction_pin: gpio.27
        step_pin: gpio.26
    motor1:
      null_motor:
  z:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 300.000
    soft_limits: false
    motor0:
      limit_all_pin: gpio.34:low
      hard_limits: false
      stepstick:
        direction_pin: gpio.33
        step_pin: gpio.32
    motor1:
      null_motor:

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5

control:

coolant:
  mist_pin: gpio.21
  delay_ms: 1000.000

probe:
  pin: gpio.22:pu
  check_mode_start: false

macros:
  startup_line0: 
  startup_line1: 
  macro0: 
  macro1: 
  macro2: 
  macro3: 

start:
  must_home: true
  check_limits: true
  deactivate_parking: false

user_outputs:

PWM:
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speeds: 0=0.0% 1000=100.0%
  output_pin: gpio.25
  enable_pin: gpio.4
  disable_with_s0: false
  s0_with_disable: false
  pwm_hz: 5000
arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
```

MitchBradley, 2021-12-27:
I reran the automatic converter with your file and it worked.  I looked at the build logs and discovered that something went wrong at github.  When the converter was trying to fetch one of the library repositories, the operation failed halfway through the repo pull.  It worked just fine later.

whins, 2021-12-27:
@MitchBradley  Thank you very match!!!

closed 2021-12-27 by whins