bdring/Grbl_Esp32 #1026: run jobs from sd card $fm error

Althafzahra, 2022-01-03:
how to solve the $fm error when using the universal Gcode sender software?  even though i saw on youtube barton dring was running normally
[image]

bdring, 2022-01-03:
I suggest you use the next version of Grbl_ESP32, which is called FluidNC

https://github.com/bdring/FluidNC

Althafzahra, 2022-01-04:
I tried to install but the error sir
[image]

MitchBradley, 2022-01-04:
According to an internet search for the error "DLL load failed ...", the problem can be caused by an old 32-bit Python installation.  You could try uninstalling old Python installations.

Another possibility is that you might be running on a 32-bit PC instead of a 64-bit one.

Althafzahra, 2022-01-04:
> According to an internet search for the error "DLL load failed ...", the problem can be caused by an old 32-bit Python installation. You could try uninstalling old Python installations.
> 
> Another possibility is that you might be running on a 32-bit PC instead of a 64-bit one.

I don't use python software, sir, only Arduino. does the serial port need to be removed first sir?