bdring/Grbl_Esp32 #1042: setting 0-5v laser power

Althafzahra, 2022-01-08:
Excuse me sir, I want to ask how to set 0-5v power for the laser, here I try m4s20-50 on the laser output still outputs 3v, should be for s50 = 2.5v, what should I do. thank you

[image]
Lightburn
[image]

bdring, 2022-01-08:
The ESP32 is a 3.2v device. It cannot output 5v.

Althafzahra, 2022-01-08:
> The ESP32 is a 3.2v device. It cannot output 5v.

> The ESP32 is a 3.2v device. It cannot output 5v.

when mode $spindle/type=laser can I set pwm for co2 laser power 20% - 100% sir?

bdring, 2022-01-08:
I think so, but I don't remember how.  ;-)

You should upgrade to FluidNC. That has a very robust power mapping feature.