bdring/Grbl_Esp32 #1057: Problems Compiling Firmware

Vladilenka, 2022-01-20:
Can't compile :-(  . Through arduino first comes the error #include <FreeRTOS.h  .I commented out this line, but the error continues . 

```
fatal error: WebSocketsServer.h: No such file or directory
compilation terminated.
```

### Tried with platformio... 

```
compilation terminated.
In file included from Grbl_Esp32/src/Grbl.h:28:0,
                 from Grbl_Esp32/src/GCode.cpp:25:
C:/.platformio/packages/framework-arduinoespressif32@3.10004.210126/cores/esp32/Arduino.h:185:26: fatal error: pins_arduino.h: No such file or directory

**********************************************************************
* Looking for pins_arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pins_arduino.h"
* Web  > https://platformio.org/lib/search?query=header:pins_arduino.h
*
**********************************************************************

compilation terminated.
*** [.pio\build\release\src\src\CoolantControl.cpp.o] Error 1
*** [.pio\build\release\src\src\GCode.cpp.o] Error 1
In file included from Grbl_Esp32/src/Grbl.h:28:0,
                 from Grbl_Esp32/src/Grbl.cpp:21:
C:/.platformio/packages/framework-arduinoespressif32@3.10004.210126/cores/esp32/Arduino.h:185:26: fatal error: pins_arduino.h: No such file or directory

**********************************************************************
* Looking for pins_arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pins_arduino.h"
* Web  > https://platformio.org/lib/search?query=header:pins_arduino.h
*
**********************************************************************

compilation terminated.
*** [.pio\build\release\src\src\Grbl.cpp.o] Error 1
In file included from Grbl_Esp32/src/Grbl.h:28:0,
                 from Grbl_Esp32/src/Jog.cpp:24:
C:/.platformio/packages/framework-arduinoespressif32@3.10004.210126/cores/esp32/Arduino.h:185:26: fatal error: pins_arduino.h: No such file or directory

**********************************************************************
* Looking for pins_arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pins_arduino.h"
* Web  > https://platformio.org/lib/search?query=header:pins_arduino.h
*
**********************************************************************

compilation terminated.
In file included from Grbl_Esp32/src/Config.h:40:0,
                 from Grbl_Esp32/src/I2SOut.cpp:43:
C:/.platformio/packages/framework-arduinoespressif32@3.10004.210126/cores/esp32/Arduino.h:185:26: fatal error: pins_arduino.h: No such file or directory

**********************************************************************
* Looking for pins_arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pins_arduino.h"
* Web  > https://platformio.org/lib/search?query=header:pins_arduino.h
*
**********************************************************************

In file included from Grbl_Esp32/src/Grbl.h:28:0,
                 from Grbl_Esp32/src/Limits.cpp:28:
C:/.platformio/packages/framework-arduinoespressif32@3.10004.210126/cores/esp32/Arduino.h:185:26: fatal error: pins_arduino.h: No such file or directory

**********************************************************************
* Looking for pins_arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pins_arduino.h"
* Web  > https://platformio.org/lib/search?query=header:pins_arduino.h
*
**********************************************************************

compilation terminated.
compilation terminated.
In file included from Grbl_Esp32/src/Motors/../Grbl.h:28:0,
                 from Grbl_Esp32/src/Motors/Motors.h:32,
                 from Grbl_Esp32/src/Motors/Motor.h:32,
                 from Grbl_Esp32/src/Motors/Dynamixel2.h:71,
                 from Grbl_Esp32/src/Motors/Dynamixel2.cpp:27:
C:/.platformio/packages/framework-arduinoespressif32@3.10004.210126/cores/esp32/Arduino.h:185:26: fatal error: pins_arduino.h: No such file or directory

**********************************************************************
* Looking for pins_arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pins_arduino.h"
* Web  > https://platformio.org/lib/search?query=header:pins_arduino.h
*
**********************************************************************

compilation terminated.
*** [.pio\build\release\src\src\Jog.cpp.o] Error 1
*** [.pio\build\release\src\src\I2SOut.cpp.o] Error 1
*** [.pio\build\release\src\src\Limits.cpp.o] Error 1
*** [.pio\build\release\src\src\Motors\Dynamixel2.cpp.o] Error 1
In file included from Grbl_Esp32/src/Grbl.h:28:0,
                 from Grbl_Esp32/src/MotionControl.cpp:25:
C:/.platformio/packages/framework-arduinoespressif32@3.10004.210126/cores/esp32/Arduino.h:185:26: fatal error: pins_arduino.h: No such file or directory

**********************************************************************
* Looking for pins_arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pins_arduino.h"
* Web  > https://platformio.org/lib/search?query=header:pins_arduino.h
*
**********************************************************************

compilation terminated.
*** [.pio\build\release\src\src\MotionControl.cpp.o] Error 1
```

### Meow. What is my problem :-( ?

P.S: The text was translated through Google translator, since I do not know English...

MitchBradley, 2022-01-20:
Have you tried FluidNC. https://github.com/bdring/FluidNC  ? It is much better than Grbl_Esp32 and does not require recompiling. Just download an installer from https://github.com/bdring/FluidNC/releases , install on your ESP32 and configure with a text file. 

We have stopped supporting grbl_esp32 and are spending all our effort now on fluidNC

Vladilenka, 2022-01-20:
**HOORAY! I did it through FluidNC! Thank you :-***

closed 2022-01-20 by Vladilenka