bdring/Grbl_Esp32 #1067: Translate Machine.h for 4 axis external board

karlosvontrapp, 2022-01-25:
[4axis_external_driver.txt](https://github.com/bdring/Grbl_Esp32/files/7932998/4axis_external_driver.txt)
This issue will automatically convert a Grbl_Esp32 "machine.h" file to FluidNC YAML format.

Change YOUR NAME in the title to some string that identifies your machine configuration.

Attach your Machine.h file here - but first change the .h extension to .txt because GitHub will not allow you to attach a file with a .h extension.  Submit the issue, and after a few minutes, a new comment will appear with the FluidNC YAML file.

abd97khateeb, 2024-05-13:
With which program should I upload the configuration file to esp32?

MitchBradley, 2024-05-13:
Don't ask questions about FluidNC on the Grbl_Esp32 issues area.  Read the FluidNC wiki for the information you need, and if you have more questions, ask on Discord or file an issue on the FluidNC github repo.