bdring/Grbl_Esp32 #1069: Hi 😊 It's work perfectly 👍

Tunisiam6, 2022-01-26:
Hi everyone , I'm trying to build 2 axis plasma table . for testing electronics and software it's working perfectly 👍.
For design and gcode I'm using Fusion 360 . Just a small modification in post processor (grbl for plasma .cps ) ,adding a relay . changing tool M6T100 (plasma torche tool number is 100 for me ) next using M3 S1000 =plasma on / M5 = plasma off .  Its working 😁. Thank you so much 🥰 well done .

ggallant571, 2022-01-26:
I am interested in your project. Can you supply some details? Did you already have the plasma cutter? How does Fusion 360 relate to grbl?

Tunisiam6, 2022-01-26:
> I am interested in your project. Can you supply some details? Did you already have the plasma cutter? How does Fusion 360 relate to grbl?

Hi, I'm working on mechanical structure right now when I finish it I will send you a video 😊... Autodesk fusion is a very good software in CAD & CAM ,I create new machine "plasma cutter" then new tool "plasma torche" . As poste processor I found a grbl compatible  cps  on GitHub ,  defining feed rate ,M3 to on plasma M5 off ,M6Txxx to change tool ... Everything working fine . 
I'm using esp32 -wifi - 16 giga sd card  , relay ,2 steppers motor .

https://photos.app.goo.gl/pjN9NdegeGofrXQw7

Tunisiam6, 2022-01-26:
Grbl for fusion 360  (add custom Post processor whe you define your machine ) tested 👍

https://github.com/jgreat/fusion360-post-grbl-plasma

MitchBradley, 2022-01-26:
We are not adding new features to Grbl_Esp32.  We have a new version[ FluidNC](https://github.com/bdring/FluidNC) that is much better.  All of our ongoing work is on FluidNC.