bdring/Grbl_Esp32 #1080: Problems Compiling Firmware

shehzad5032, 2022-02-07:
Please answer the following questions:

Have you read the [wiki regarding how to compile](https://github.com/bdring/Grbl_Esp32/wiki/Compiling-the-firmware)?y

What version of the Arduino IDE are you using? 1.8.19

What version (commit date) of the [Arduino core for the ESP32](https://github.com/espressif/arduino-esp32) are you using? 

Are you using the master branch of Grbl_ESP32?y

Have you made any edits or configuration changes (list them) to the  firmware?n

Please paste the compiler error text here:C:\Users\shehzad\Desktop\Grbl_Esp32-main\Grbl_Esp32\src\I2SOut.cpp:54:10: fatal error: FreeRTOS.h: No such file or directory
 #include <FreeRTOS.h>
          ^~~~~~~~~~~~
compilation terminated.
Multiple libraries were found for "SD.h"
 Used: C:\Users\xxxxxxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\SD
 Not used: C:\Program Files (x86)\Arduino\libraries\SD
Multiple libraries were found for "WiFi.h"
 Used: C:\Users\xxxxxxxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
exit status 1
Error compiling for board ESP32 Dev Module.

closed 2022-02-07 by shehzad5032

MitchBradley, 2022-02-07:
You should consider switching to FluidNC.  Grbl_Esp32 is no longer actively supported.  We are spending all of our time on FluidNC now.

hugs, 2022-03-20:
I ran into this same error. In the Arduino Boards Manager, I was running esp version 2.0.2. I solved this issue by downgrading the esp version down 1.0.6. No other changes required to get a clean compile.

I'll look into upgrading to FluidNC when I can, but I'm running custom kinematics (deltabots) and porting all that over to FluidNC is going to take some brain cycles.