bdring/Grbl_Esp32 #1082: Translate Machine.h for SStrong

SS-mill, 2022-02-08:
This issue will automatically convert a Grbl_Esp32 "machine.h" file to FluidNC YAML format.

Change YOUR NAME in the title to some string that identifies your machine configuration.

Attach your Machine.h file here - but first change the .h extension to .txt because GitHub will not allow you to attach a file with a .h extension.  Submit the issue, and after a few minutes, a new comment will appear with the FluidNC YAML file.
[machine.txt](https://github.com/bdring/Grbl_Esp32/files/8020286/machine.txt)

MitchBradley, 2022-02-08:
Getting closer.  Now it is unhappy with the spindle type which should be written as SpindleType::PWM .

In general, if the machine.h file will not compile with the latest Grbl_Esp32 code, the converter won't handle it either.

MitchBradley, 2022-02-08:
You will be better off starting with one of the machine.h files that are in the Machines/ folder in the main branch, instead of some old one from some other place.