bdring/Grbl_Esp32 #1111: Translate Machine.h for tulip

easytarget, 2022-02-21:
This issue will automatically convert a Grbl_Esp32 "machine.h" file to FluidNC YAML format.

Change YOUR NAME in the title to some string that identifies your machine configuration.

Attach your Machine.h file here - but first change the .h extension to .txt because GitHub will not allow you to attach a file with a .h extension.  Submit the issue, and after a few minutes, a new comment will appear with the FluidNC YAML file.

[machine.txt](https://github.com/bdring/Grbl_Esp32/files/8112367/machine.txt)

MitchBradley, 2022-02-21:
You need to say 
#define SPINDLE_TYPE            SpindleType::PWM
instead of
#define SPINDLE_TYPE            SPINDLE_TYPE_PWM

easytarget, 2022-02-21:
:-) thankyou, I figured my old config is a very old config, there may be other things lurking in it.. so I'm bitingthe bullet and doing a from-scratch re-config. It's good practice for me anyway

closed 2022-02-21 by easytarget