bdring/Grbl_Esp32 #1137: no grbl_notifyf/grbl_msg_sendf messages during the task xTaskCreatePinnedToCore

cosmok82, 2022-03-25:
Hi at all,

I'd like to add the sensor as5600 to my personal scara kinematics, but when I call through the serial a grbl_notifyf or a grbl_msg_sendf I have no printing messages on the serial monitor. Do you know what's the problem? I attached just an example:

```
static TaskHandle_t as5600UpdateTaskHandle;

void as5600Update(void* pvParameters) {
    TickType_t       xLastWakeTime;
    const TickType_t xDisplayFrequency = 100;                  // in ticks (typically ms)
    xLastWakeTime                      = xTaskGetTickCount();  // Initialise the xLastWakeTime variable with the current time.

    vTaskDelay(1000);
    
    while (true) {
      TCA9548A(2);
      ReadRawAngle();

      grbl_notifyf("AS5600 sensor", "Angle: %4.2f \r\n", degAngle);
      grbl_msg_sendf(CLIENT_SERIAL, MsgLevel::Verbose, "Angle: %4.2f", degAngle);

      vTaskDelayUntil(&xLastWakeTime, xDisplayFrequency);
    }
}

void as5600_init() {
    // Initialising the UI will init the display too.
    grbl_msg_sendf(CLIENT_SERIAL, MsgLevel::Info, "Init AS5600 SDA:%s SCL:%s", pinName(I2C_SDA_PIN), pinName(I2C_SCL_PIN));
    
    xTaskCreatePinnedToCore(as5600Update,                // task
                            "as5600UpdateTask",          // name for task
                            4096,                        // size of task stack
                            NULL,                        // parameters
                            1,                           // priority
                            &as5600UpdateTaskHandle,     //
                            CONFIG_ARDUINO_RUNNING_CORE  // must run the task on same core
                                                         // core
    );
}
```

as5600_init(); it was added to Grbl.cpp and it works pretty well, but I don't read any status message on serial after the initialization from as5600 task as expected. I'd like to evaluate it in debug mode at least, or in verbose mode if possible, during the development.

MitchBradley, 2022-03-25:
vTaskDelayUntil(&xLastWakeTime, xDisplayFrequency);

That asks it to wait until some time in the past.

cosmok82, 2022-03-25:
> vTaskDelayUntil(&xLastWakeTime, xDisplayFrequency);
> 
> That asks it to wait until some time in the past.

Sorry, but i don't understand how it could be useful. For example I tried with a blinking on a GPIO and it work perfectly, why not on a serial?

MitchBradley, 2022-03-25:
Try sending $Message/Level=Verbose from the serial terminal.  Either that or do the sendf with MessageLevel::Info

100ms is a little fast for messages.  I would change it to 500

cosmok82, 2022-03-25:
> 

I changed the code to this:
```

static TaskHandle_t as5600UpdateTaskHandle;

void as5600Update(void* pvParameters) {
    
    while (true) {
      TCA9548A(2);
      ReadRawAngle();

      grbl_msg_sendf(CLIENT_SERIAL, MsgLevel::Info, "Angle: %4.2f \r\n", degAngle);
      delay(1000);

    }
}

void as5600_init() {
    // Initialising the UI will init the display too.
    grbl_msg_sendf(CLIENT_SERIAL, MsgLevel::Info, "Init AS5600 SDA:%s SCL:%s", pinName(I2C_SDA_PIN), pinName(I2C_SCL_PIN));
    
    xTaskCreatePinnedToCore(as5600Update,                // task
                            "as5600UpdateTask",          // name for task
                            4096,                        // size of task stack
                            NULL,                        // parameters
                            1,                           // priority
                            &as5600UpdateTaskHandle,     //NULL
                            CONFIG_ARDUINO_RUNNING_CORE  // must run the task on same core
                                                         // core
    );
}
```

But nothing, it doesn't work. Could be a problem with tasks?
The serial lib is on a different task, maybe could be this.

MitchBradley, 2022-03-25:
Did you actually call as5600_init() from somewhere in the main initialization code?

cosmok82, 2022-03-25:
> Did you actually call as5600_init() from somewhere in the main initialization code?

Of course, as I said in the first message "_**as5600_init(); it was added to Grbl.cpp and it works pretty well**_"

This is the output:

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8

[MSG:Init AS5600 SDA:GPIO(21) SCL:GPIO(22)]
[MSG:Grbl_ESP32 Ver 1.3a Date 20211103]
 
```

cosmok82, 2022-03-25:
> FluidNC

Sorry Mitch, maybe I get wrong, but I'm on FluidNC https://github.com/bdring/FluidNC

MitchBradley, 2022-03-25:
I added this code to Main.cpp as a test:
```
void as5600Update(void* pvParameters) {
    log_info("Hello from task");
    while (true) {
        log_info("Ping from task");
        delay(1000);
    }
}

void as5600_init() {
    // Initialising the UI will init the display too.
    log_info("Creating task");

    xTaskCreatePinnedToCore(as5600Update,                // task
                            "as5600UpdateTask",          // name for task
                            4096,                        // size of task stack
                            NULL,                        // parameters
                            1,                           // priority
                            &as5600UpdateTaskHandle,     //NULL
                            CONFIG_ARDUINO_RUNNING_CORE  // must run the task on same core
                                                         // core
    );
}
```
and added this as the last line of setup()
```
    as5600_init();
```

It worked.

[image]

That is all the help I can provide.

closed 2022-03-25 by MitchBradley

cosmok82, 2022-03-25:
> setup(

Sorry but this is Grbl_Esp32, not FluidNC. Moreover, I need it in my custom, not in setup, so it isn't solved. There is a problem in my opinion. What terminal do you use? I tried on the Arduino monitor and I got nothing.

[image]

[image]

cosmok82, 2022-03-25:
> > setup(
> 
> Sorry but this is Grbl_Esp32, not FluidNC. Moreover, I need it in my custom, not in setup, so it isn't solved. There is a problem in my opinion. What terminal do you use? I tried on the Arduino monitor and I got nothing.
> 
> [image]
> 
> [image]

The problem has been got around not solved, in fact, it doesn't work in the correct path, it's a different scenario in your case. To be solved have to be respected the path!

MitchBradley, 2022-03-25:
I am not willing to support Grbl_Esp32 any more.  My code just proves that serial output from tasks work in FluidNC.  I put it in Main.cpp because that was the easiest place for me to test it. There is no inherent reason why it would not work if the code is in some other file, so long as the initialization routine is called at an appropriate time after the serial port and I2C has been set up.

I tried it with an experimental web-based terminal, with FluidTerm, and with TeraTermPro.  It worked in all cases.

At this point you are on your own unless someone else is willing to help you.  Any time that I spend helping people with programming problems or teaching how the code base is structured is time that I am not spending improving the FluidNC code.  I choose to do the latter.

cosmok82, 2022-03-25:
> I am not willing to support Grbl_Esp32 any more. My code just proves that serial output from tasks work in FluidNC. I put it in Main.cpp because that was the easiest place for me to test it. There [...]

I understand your goals, but giving a petty solution isn't giving a solution to the proposed problem, it squeaks too much and doesn't even come close to the solution. Anyway thanks for your time, but honestly it isn't solved at all. If I said otherwise it wouldn't be professional.

cosmok82, 2022-03-26:
I reached the genuine solution! Take into account that the HW is mine and in development.

The first problem was the HW.

SCL and SDA pins has been swapped and the TCA9548A had no communication. Moreover, the A0 A1 A2 pins of TCA9548A was flotting, I checked the HW and I found another lack.
These pins had to be conneced to the GND of the ESP32 to set the 0x70 address.

The last problem was the SW.

On the SW, the Wire library had to be configurated ad in the screenshot (that's my configuration, in your case could changes), at the same time the magnet had to be on sensor to receive a signal.

And, last but not least, I suggest a clean of software compiled before upload it, could be useful just in case of multiple modification, easy! 
I think that's all.

Now it's SOLVED. 

It's my proprietary project, I know, but the thought to leave a petty solution to the community left a bitter taste in my mouth.

[image]