bdring/Grbl_Esp32 #1158: Homing offsetting Y axis

Mamoonpk, 2022-04-12:
Hi
I have gone through other issues raised for same subject. This issue is little different.
I am using 6 pack universal board, with dm556T external drivers. Having 2 ganged motors on y axis. Machine bed size is arround 475mm x 475mm. Its a custom build.
I used it so far with arduino with out issues and right now wanted to update to esp32 

After setup I have noticed that if I jogged machine on y axis back and forth it reaches to same point every time. But when I home the machine, it cause huge offset on my y axis, which later recovers in auto squaring. But it puts alot of stress initially to gantry before squaring.

My question is that if I can jog machine on y axis and reaching to same point again  and again then why while homing both stepper motors runs with different speeds? I believe that is the only reason causing offset if my axis was initially squared.

In my opinion, on homing, machine should consider the axis as it is already square. If there is any offset, it should square with tiny adjustment at the end.

Why both ganged motors at y axis runs with different speeds while homing? Is there anything that I am missing in my setup?

MitchBradley, 2022-04-12:
Try it with FluidNC.  We have made a lot of improvements in the homing code.  FluidNC is so far ahead of Grbl_Esp32 now that we are no longer supporting Grbl_Esp32.

closed 2022-04-12 by MitchBradley

Mamoonpk, 2022-04-12:
Further details.
I am using nema 23 on y access. With 6400 steps / rev settings on both drivers
Grbl 1.3a
Ugs 2.0.10

MitchBradley, 2022-04-12:
Giving us additional details will do you no good.  As I said, Grbl_Esp32 is no longer supported.  If you switch over to FluidNC we can and will help you.