bdring/Grbl_Esp32 #1177: Cnc_3Axes_Ysquarred_michel

faumi78, 2022-04-25:
pragma once
// clang-format off

/*
	3axis_xyx.h
	Part of Grbl_ESP32

	CNC 3 Axes avec 2 moteurs en Y et 2 switch limits separés.
	Le homing des 3 axes est séparé et les axes Y gauche et droit se font séparément sur les derniers milimètres.
	La config de la machine n'est pas intégré a ce fichier.
*/

#define MACHINE_NAME            "Cnc_3Axes_Ysquarred_michel"

# define  DEFAULT_HOMING_CYCLE_0  bit (Z_AXIS)
# define  DEFAULT_HOMING_CYCLE_1  bit (X_AXIS)
# define  DEFAULT_HOMING_CYCLE_2  bit (Y_AXIS)
#define DEFAULT_HOMING_SQUARED_AXES bit(Y_AXIS)

#define X_STEP_PIN              GPIO_NUM_14   // labeled X
#define X_DIRECTION_PIN         GPIO_NUM_12  // labeled X
#define Y_STEP_PIN              GPIO_NUM_33  // labeled Y
#define Y_DIRECTION_PIN         GPIO_NUM_25  // labeled Y
#define Y2_STEP_PIN             GPIO_NUM_32  // labeled Y2
#define Y2_DIRECTION_PIN        Y_DIRECTION_PIN  // labeled Y2
#define Z_STEP_PIN              GPIO_NUM_26  // labeled Z
#define Z_DIRECTION_PIN         GPIO_NUM_27  // labeled Z
#define STEPPERS_DISABLE_PIN    GPIO_NUM_13

#define SPINDLE_TYPE            SpindleType::PWM
#define SPINDLE_OUTPUT_PIN      GPIO_NUM_17  // labeled SpinPWM

#define X_LIMIT_PIN             GPIO_NUM_4
#define Y_LIMIT_PIN             GPIO_NUM_2
#define Y2_LIMIT_PIN            GPIO_NUM_16
#define Z_LIMIT_PIN             GPIO_NUM_15

// Set $Homing/Cycle0=X and $Homing/Cycle=XY

#define PROBE_PIN               GPIO_NUM_22  // labeled Probe
#define CONTROL_SAFETY_DOOR_PIN GPIO_NUM_35  // labeled Door,  needs external pullup
//#define CONTROL_RESET_PIN       GPIO_NUM_34  // labeled Reset
#define CONTROL_FEED_HOLD_PIN   GPIO_NUM_36  // labeled Hold
//#define CONTROL_CYCLE_START_PIN GPIO_NUM_39  // labeled Start

#define INVERT_CONTROL_PIN_MASK B1100 // Pour la securité utilise une boucle fermée c'est a dire un contact NC mais la carte boitier inverse le signal
// #define INVERT_CONTROL_PIN_MASK  (0<<CONTROL_PIN_INDEX_SAFETY_DOOR) autre solution pour inverser 

// #define VFD_DEBUG_MODE

MitchBradley, 2022-04-25:
Please stop trying the same wrong thing over and over

MitchBradley, 2022-04-25:
I will translate the file for you if you just stop resubmitting the issue.

MitchBradley, 2022-04-25:
Look at issue #1182 for the translated file for your machine