bdring/Grbl_Esp32 #1219: Translate Machine.h for XproV5 MF70

am3ist3r, 2022-06-09:
This issue will automatically convert a Grbl_Esp32 machine definition file to FluidNC YAML format.

Change YOUR NAME in the title to some string that identifies your machine configuration.

IMPORTANT: Change only the YOUR NAME part of the title, not the entire title.  If you change the entire title, GitHub will not run the converter program.
[Xpro090622.txt](https://github.com/bdring/Grbl_Esp32/files/8873028/Xpro090622.txt)

Attach your machine definition file here - but first change the .h extension to .txt because GitHub will not allow you to attach a file with a .h extension.  Submit the issue, and after a few minutes, a new comment will appear with the FluidNC YAML file.

IMPORTANT: Attach the file - do not paste the file text into the issue.  If you paste, the converter will not work.

MitchBradley, 2022-06-09:
You must attach the Machine.h file, not the settings.

MitchBradley, 2022-06-10:
The file that you attached contained the NVS settings.  The translator works on Machine.h files, not lists of settings.