bdring/Grbl_Esp32 #1244: Errors with Arduino-ESP32 ver 2.0-...

slresour, 2022-07-07:
Errors with Arduino-ESP32 ver 2.0-...
"arduinoWebSockets\src\WebSockets.cpp:42:10: fatal error: hwcrypto/sha.h: No such file or directory"
"Grbl_Esp32-main\Grbl_Esp32\src\I2SOut.cpp:54:10: fatal error: FreeRTOS.h: No such file or directory"

With Arduino-ESP32 ver 1.8.6 no problems.

slresour, 2022-07-08:
In my opinion FluidNC is worse than Grbl-ESP32. 
For example: 
   $$ 
   $10=1 Ok. 
Is that all? How do I change the parameters by bluetuth?

Avataar120, 2022-07-13:
FluidNC is using YAML configuration file
Everything is explained in the very detailed wiki https://github.com/bdring/FluidNC/wiki/FluidNC-Config-File-Overview

MitchBradley, 2022-07-13:
FluidNC is much more configurable than Grbl_Esp32.  The configuration is now in a hierarchical tree-structured file, whose parameters have descriptive names and human-readable values instead of meaningless number "names" and numerical-only values.  Aspects of Grbl_Esp32 that required recompilation to change can now be configured with a text file, so that few users need to set up and maintain a compilation environment.  Releases can be used as-is, with a one-click installation process that preserves the user's configuration.

To upload a YAML configuration file over Bluetooth, you can use FluidTerm - included in the release package - to connect via Bluetooth, then use FluidTerm's CTRL-U upload command to put your configuration file into the ESP32 FLASH filesystem.

Existing Grbl_Esp32 compile-time configurations (the Machine.h file) can be auto-converted to FluidNC YAML files by submitting a Grbl_Esp32 issue using the "Translate Machine.h File to FluidNC".