bdring/Grbl_Esp32 #1249: Translate Machine.h for JJGShapeOKO

jjgerber01, 2022-07-25:
This issue will automatically convert a Grbl_Esp32 machine definition file to FluidNC YAML format.

Change YOUR NAME in the title to some string that identifies your machine configuration.

IMPORTANT: Change only the YOUR NAME part of the title, not the entire title.  If you change the entire title, GitHub will not run the converter program.

Attach your machine definition file here - but first change the .h extension to .txt because GitHub will not allow you to attach a file with a .h extension.  Submit the issue, and after a few minutes, a new comment will appear with the FluidNC YAML file.

IMPORTANT: Attach the
[6_pack_Shapeoko3_stepstick_XYYZ_1_SW_2_SW_3_Relay_4_5V_5_Relay.txt](https://github.com/bdring/Grbl_Esp32/files/9182552/6_pack_Shapeoko3_stepstick_XYYZ_1_SW_2_SW_3_Relay_4_5V_5_Relay.txt)
 file - do not paste the file text into the issue.  If you paste, the converter will not work.

MitchBradley, 2022-07-25:
FYI, that filename will not work as the name of the config file on the FluidNC system. The SPIFFS filesystem has a 32-character limit on the name length.

jjgerber01, 2022-07-25:
Thank you. I have renamed it "config.yaml"

MitchBradley, 2022-07-25:
I suggest a simple version number too, in case you need to tweak some settings.

jjgerber01, 2022-07-25:
Ok, good idea. Thank you