bdring/Grbl_Esp32 #1251: Translate Machine.h for hoverCNC

HoverClub, 2022-07-30:
This issue will automatically convert a Grbl_Esp32 machine definition file to FluidNC YAML format.

Change YOUR NAME in the title to some string that identifies your machine configuration.

IMPORTANT: Change only the YOUR NAME part of the title, not the entire title.  If you change the entire title, GitHub will not run the converter program.

Attach your machine definition file here - but first change the .h extension to .txt because GitHub will not allow you to attach a file with a .h extension.  Submit the issue, and after a few minutes, a new comment will appear with the FluidNC YAML file.

IMPORTANT: Attach the file - do not paste the file text into the issue.  If you paste, the converter will not work.
[GrblSettings.txt](https://github.com/bdring/Grbl_Esp32/files/9226120/GrblSettings.txt)

MitchBradley, 2022-07-30:
It is not the $ setting that you have to attach, it is the C language  .h file that tells, among other things, which pins are used for things like step and direction.

HoverClub, 2022-07-30:
Thanks, I realised that two seconds after attaching the file!