bdring/Grbl_Esp32 #1253: I start when the y-axis limit is triggered

lfmmd, 2022-08-03:
I found that the serial port is there in the first few seconds of boot, after that it will not exist, I don't know if it's a problem with the board or the software my board is cheap   http://www.grbl.cc/9999999-2   ESP32 GRB

MitchBradley, 2022-08-03:
There is no documentation on that board so we cannot determine which pins it uses.  ESP32 has some pins with special meanings at boot time.  If one of those is used as the y-axis limit, that would cause problems.  See https://github.com/bdring/FluidNC/wiki/ESP32-Pin-Reference#usable-io-pins and https://github.com/bdring/FluidNC/wiki/ESP32-Pin-Reference#dont-use-or-not-recommended

I am closing this ticket now because we cannot debug hardware that is not documented.  The use of our software is governed by a GPL license that requires that the seller provide source code for any modifications that they have made.  The machine configuration file that defines the pin usage is such a modification.  If they do not provide it, their use of our software could be illegal.

closed 2022-08-03 by MitchBradley

lfmmd, 2022-08-04:
Thank you for your reply