bdring/Grbl_Esp32 #1267: Translate Machine.h for LukiElectronics custom Control Board

LukiElectronics, 2022-08-19:
This issue will automatically convert a Grbl_Esp32 machine definition file to FluidNC YAML format.

Change YOUR NAME in the title to some string that identifies your machine configuration.

IMPORTANT: Change only the YOUR NAME part of the title, not the entire title.  If you change the entire title, GitHub will not run the converter program.

Attach your machine definition file here - but first change the .h extension to .txt because GitHub will not allow you to attach a file with a .h extension.  Submit the issue, and after a few minutes, a new comment will appear with the FluidNC YAML file.

IMPORTANT: Attach the file - do not paste the file text into the issue.  If you paste, the converter will not work.
[machine.txt](https://github.com/bdring/Grbl_Esp32/files/9383965/machine.txt)

PS: Third time's the charm? Hopefully ...

closed 2022-08-19 by LukiElectronics

MitchBradley, 2022-08-19:
Here is the error message from the action runner that failed to translate your file

[image]

I think that UART_NUM_2 is not defined in some versions of the framework, but since UART_NUM_n resolves to just n, you could try just saying 2.

LukiElectronics, 2022-08-19:
Thank you, i will try that, and if it works this will save me quite a bit effort.