bdring/Grbl_Esp32 #1268: Translate Machine.h for LukiElectronics custom Control Board

LukiElectronics, 2022-08-19:
This issue will automatically convert a Grbl_Esp32 machine definition file to FluidNC YAML format.

Change YOUR NAME in the title to some string that identifies your machine configuration.

IMPORTANT: Change only the YOUR NAME part of the title, not the entire title.  If you change the entire title, GitHub will not run the converter program.

Attach your machine definition file here - but first change the .h extension to .txt because GitHub will not allow you to attach a file with a .h extension.  Submit the issue, and after a few minutes, a new comment will appear with the FluidNC YAML file.

IMPORTANT: Attach the file - do not paste the file text into the issue.  If you paste, the converter will not work.
[machine.txt](https://github.com/bdring/Grbl_Esp32/files/9384383/machine.txt)

PS: Sorry for the fourth issue, i did the change you suggested hopefully this works.

MitchBradley, 2022-08-19:
TMC2209 has only two address pins so addr: 4 is not useful.

LukiElectronics, 2022-08-19:
I know, and already changed that, but I am trying to run the two drivers of the Y axis with the same address in order to only need one Uart interface of the Esp and no analog switching between two networks.

closed 2022-08-19 by LukiElectronics