bdring/Grbl_Esp32 #1270: Translate Machine.h for ZBaitu (81x46)

mmeulstee, 2022-08-28:
This issue will automatically convert a Grbl_Esp32 machine definition file to FluidNC YAML format.

Change YOUR NAME in the title to some string that identifies your machine configuration.

IMPORTANT: Change only the YOUR NAME part of the title, not the entire title.  If you change the entire title, GitHub will not run the converter program.

Attach your machine definition file here - but first change the .h extension to .txt because GitHub will not allow you to attach a file with a .h extension.  Submit the issue, and after a few minutes, a new comment will appear with the FluidNC YAML file.

IMPORTANT: Attach the file - do not paste the file text into the issue.  If you paste, the converter will not work.
[M81_Configuration.txt](https://github.com/bdring/Grbl_Esp32/files/9440085/M81_Configuration.txt)

MitchBradley, 2022-08-28:
The converter converts .h files that define thinks like the controller pinout.  The file that you attached is a $$ settings file that defines things like steps per mm and rates.  The converter cannot handle $$ setttings.

mmeulstee, 2022-08-29:
How do I get that file if I only have a precompiled grbl_esp32

MitchBradley, 2022-08-29:
Ask whoever provided the precompiled file to give you the config file - or search their web site or github repos for one.  If they refuse, they are in violation of Grbl_Esp32's GPL license terms.  Another possibility, for vendors who are good at hiding, is to look through old "Translate Machine" issues to see if someone has already performed a translation for a board like yours.  Such people may have succeeded in finding a config file or in creating one from other information like schematics or reverse engineering.

mmeulstee, 2022-08-29:
Ok gonna check it out

mmeulstee, 2022-08-30:
Asked twice for the files now and still no response.
Anyway i found some information tough:

[MSG:Using machine:LASER_ENGRAVER]
[MSG:Axis count 3]
[MSG:RMT Steps]
[MSG:Macro Pin 0 GPIO(0)]
[MSG:Init Motors]
[MSG:Global stepper disable pin:GPIO(12)]
[MSG:X  Axis Standard Stepper Step:GPIO(14) Dir:GPIO(27) Disable:None Limits(0.000,810.000)]
[MSG:Y  Axis Standard Stepper Step:GPIO(26) Dir:GPIO(25) Disable:None Limits(0.000,460.000)]
[MSG:Y2 Axis Standard Stepper Step:GPIO(33) Dir:GPIO(32) Disable:None Limits(0.000,460.000)]
[MSG:Laser spindle on Pin:GPIO(17), Enbl:None, Freq:5000Hz, Res:13bits Laser mode:On]

$Stepper/Enable/Delay=0
$Stepper/Direction/Delay=0
$Stepper/Pulse=3
$Stepper/IdleTime=250
$Stepper/StepInvert=
$Stepper/DirInvert=Y
$Stepper/EnableInvert=Off
$Limits/Invert=On
$Probe/Invert=Off
$Report/Status=1
$GCode/JunctionDeviation=0.010
$GCode/ArcTolerance=0.002
$Report/Inches=Off
$Firmware/Build=
$Limits/Soft=Off
$Limits/Hard=Off
$Homing/Enable=On
$Homing/DirInvert=XY
$Homing/Squared=
$Homing/Feed=200.000
$Homing/Seek=3000.000
$Homing/Debounce=250.000
$Homing/Pulloff=3.000
$GCode/MaxS=1000.000
$GCode/MinS=0.000
$Laser/FullPower=1000
$GCode/LaserMode=On
$GCode/Line1=
$GCode/Line0=
$Spindle/Enable/Invert=Off
$Spindle/Enable/OffWithSpeed=Off
$Coolant/Delay/TurnOn=1.000
$Spindle/Delay/SpinDown=0.000
$Spindle/Delay/SpinUp=0.000
$Spindle/PWM/Invert=Off
$Spindle/PWM/Frequency=5000.000
$Spindle/PWM/Off=0.000
$Spindle/PWM/Min=0.000
$Spindle/PWM/Max=100.000
$Spindle/Type=LASER
$X/StepsPerMm=80.000
$Y/StepsPerMm=80.000
$Z/StepsPerMm=80.000
$A/StepsPerMm=80.000
$B/StepsPerMm=80.000
$C/StepsPerMm=80.000
$X/MaxRate=50000.000
$Y/MaxRate=50000.000
$Z/MaxRate=50000.000
$A/MaxRate=50000.000
$B/MaxRate=50000.000
$C/MaxRate=50000.000
$X/Acceleration=800.000
$Y/Acceleration=800.000
$Z/Acceleration=800.000
$A/Acceleration=800.000
$B/Acceleration=800.000
$C/Acceleration=800.000
$X/Home/Mpos=0.000
$Y/Home/Mpos=0.000
$Z/Home/Mpos=0.000
$A/Home/Mpos=0.000
$B/Home/Mpos=0.000
$C/Home/Mpos=0.000
$X/MaxTravel=810.000
$Y/MaxTravel=460.000
$Z/MaxTravel=460.000
$A/MaxTravel=810.000
$B/MaxTravel=460.000
$C/MaxTravel=460.000
$X/Current/Run=0.250
$Y/Current/Run=0.250
$Z/Current/Run=0.250
$A/Current/Run=0.250
$B/Current/Run=0.250
$C/Current/Run=0.250
$X/Current/Hold=0.125
$Y/Current/Hold=0.125
$Z/Current/Hold=0.125
$A/Current/Hold=0.125
$B/Current/Hold=0.125
$C/Current/Hold=0.125
$X/Microsteps=16
$Y/Microsteps=16
$Z/Microsteps=16
$A/Microsteps=16
$B/Microsteps=16
$C/Microsteps=16
$X/StallGuard=16
$Y/StallGuard=16
$Z/StallGuard=16
$A/StallGuard=16
$B/StallGuard=16
$C/StallGuard=16

Still don't know how to implement it in a config(.h/.yaml)

bdring, 2022-08-30:
We basically need to reverse engineer the controller.

 - Does it have any limit switches connected.
 - What are the red arrow connectors for?
 - What is the blue arrow thing?
 - What type of motor drivers are they. Green arrow (A4988, etc)
 
 
[image]

mmeulstee, 2022-08-30:
[image]

Y-Limit on the board with micro switch
Motor Connections are connected like:
X and Y are used for Y-Motors
Z is used for X-Motor

Stepdrivers are 4988ET

The X-limit and Laser does share the same GND  ----- [12V-GND-PWM-Switch]

bdring, 2022-08-30:
Are the limit switches used for homing?

mmeulstee, 2022-08-30:
Yes Homing at X0 and Y0

bdring, 2022-08-30:
Are there any [MSG:  .... ] lines that show the switch pins?

mmeulstee, 2022-08-30:
[MSG:X  Axis limit switch on pin GPIO(13)]
[MSG:Y  Axis limit switch on pin GPIO(39)]
[MSG:Z  Axis limit switch on pin GPIO(34)]

bdring, 2022-08-30:
- Is The Z switch used as a second Y switch?
- Does Y home down
- Does X home to the left?

mmeulstee, 2022-08-30:
For now the Z isnt used yet (future plans)
Yes for Y and X

mmeulstee, 2022-08-30:
I guess the Y/X question was for  
positive_direction: false

mmeulstee, 2022-08-30:
Again thanks for the help and effort in this, i'm off to bed.
Will catch this tomorrow afternoon

bdring, 2022-08-30:
I suggest you use the Z switch input as the second Y. I have made the config file for that setup.

here is is.

```yaml
board: Zbaitu Controller
name: Zbaitu XYY Laser
meta: B. Dring for @mmeulstee 8/30/2022

stepping:
  engine: RMT
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
  
start:
  must_home: false
  deactivate_parking: false
  check_limits: false

axes:
  shared_stepper_disable_pin: gpio.12
  x:
    steps_per_mm: 80.000
    max_rate_mm_per_min: 50000.000
    acceleration_mm_per_sec2: 800.000
    max_travel_mm: 810
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 3000.000
      settle_ms: 250
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.13:pu:low
      hard_limits: false
      pulloff_mm: 1.000
      stepstick:
        step_pin: gpio.14
        direction_pin: gpio.27
        
  y:
    steps_per_mm: 80.000
    max_rate_mm_per_min: 50000.000
    acceleration_mm_per_sec2: 800.000
    max_travel_mm: 460.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 3000.000
      settle_ms: 250
      seek_scaler: 1.100
      feed_scaler: 1.100
      

    motor0:
      limit_neg_pin: gpio.39:low
      pulloff_mm: 1.000
      hard_limits: false
      stepstick:
        step_pin: gpio.26
        direction_pin: gpio.25:low
        
    motor1:
      limit_neg_pin: gpio.34:low
      pulloff_mm: 1.000
      hard_limits: false
      stepstick:
        step_pin: gpio.33
        direction_pin: gpio.32:low
        
spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

Laser:
  pwm_hz: 5000
  output_pin: gpio.17
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  tool_num: 0
  speed_map: 0=0.000% 1000=100.000%
  ```

bdring, 2022-08-31:
I started a wiki page for it.

http://wiki.fluidnc.com/en/hardware/3rd-party/zbaitu_laser

Let me know what changes are needed.

mmeulstee, 2022-08-31:
Works like a charm, thank you very much.
The only thing i had to change was:
[y/motor1] direction_pin: gpio.32:low -> direction_pin: gpio.32
and remove the second Y-endstop since i dont have the hardware yet.

closed 2022-08-31 by bdring