bdring/Grbl_Esp32 #1273: Translate Machine.h for YOUR NAME CNC_laser

richardwpaul, 2022-09-04:
This issue will automatically convert a Grbl_Esp32 machine definition file to FluidNC YAML format.

Change YOUR NAME in the title to some string that identifies your machine configuration.

IMPORTANT: Change only the YOUR NAME part of the title, not the entire title.  If you change the entire title, GitHub will not run the converter program.

Attach your machine definition file here - but first change the .h extension to .txt because GitHub will not allow you to attach a file with a .h extension.  Submit the issue, and after a few minutes, a new comment will appear with the FluidNC YAML file.

IMPORTANT: Attach the file - do not paste the file text into the issue.  If you paste, the converter will not work.

richardwpaul, 2022-09-04:
[CNC_Laser.txt](https://github.com/bdring/Grbl_Esp32/files/9485439/CNC_Laser.txt)

richardwpaul, 2022-09-04:
Thank you for your work and help with converting over to FluidCNC.

MitchBradley, 2022-09-04:
You will need to start s new issue with the .txt file. The boy that does the translation is too dumb to process followup comments