bdring/Grbl_Esp32 #1279: ESP32_GRBL BLE connect to iOS device

leo1988joseph, 2022-09-12:
Hi, I have an issue connecting the ESP32 to iPhone device. I enabled the ESP32 BT mode, but it doesn't shows the BT name on the device. I tried the same in an Android, I saw the BT in the list and connected to it. But in the serial monitor I typed $I and saw that the mode is "Not connected". Could anyone help?

MitchBradley, 2022-09-12:
I do not think that iOS support the ordinary Bluetooth serial profile.  iOS mostly handle Bluetooth Low Energy, AKA BLE or Bluetooth 4 devices.  Grbl_Esp32 and it successor FluidNC use the legacy Bluetooth Serial protocol over "classic" Bluetooth, not over BLE.

leo1988joseph, 2022-09-13:
So what would you suggest to move the motor via esp32-grbl combination via iOS? Any work-around? I am fairly new to controllers and boards. And I already started off with the project.

MitchBradley, 2022-09-13:
First of all, Grbl_Esp32 is no longer supported.  It has been superseded by FluidNC.

You can use WiFi to control FluidNC (or Grbl_Esp32) via the built-in WebUI user interface.  It has a Tablet tab that is optimized for small touchscreens, in addition to the normal UI tab that works well with larger screens and a mouse.

leo1988joseph, 2022-09-13:
I am looking for a way to dynamically send the x/y coordinates and make the axes move correspondingly. I have to use the native iOS language swift to pass the value. I need help regarding which file in the FluidNC can accept a variable dynamically?

MitchBradley, 2022-09-13:
You can connect using the websocket protocol to port 81 and then send GCode commands over that stream.  The send/response protocol is the same as for standard Grbl, except that the characters go over the websocket stream instead of serial.  A websocket stream can be open at the same time as the serial port.  You can also use raw TCP in the same way, via the standard Telnet port.

leo1988joseph, 2022-09-13:
Does the library or wiki have any sample code that shows this flow? Or any link that does the same for me to regenerate please?

MitchBradley, 2022-09-13:
There is no sample code in Swift for anything related to Grbl_Esp32 or FluidNC.

closed 2022-09-13 by MitchBradley