bdring/Grbl_Esp32 #1291: Translate Machine.h for MY_Laser_Translated

graham2550, 2022-09-30:
This issue will automatically convert a Grbl_Esp32 machine definition file to FluidNC YAML format.

Change YOUR NAME in the title to some string that identifies your machine configuration.

IMPORTANT: Change only the YOUR NAME part of the title, not the entire title.  If you change the entire title, GitHub will not run the converter program.

Attach your machine definition file here - but first change the .h extension to .txt because GitHub will not allow you to attach a file with a .h extension.  Submit the issue, and after a few minutes, a new comment will appear with the FluidNC YAML file.

IMPORTANT: Attach the file - do not paste the file text into the issue.  If you paste, the converter will not work.
[mpcnc_v1p2.txt](https://github.com/bdring/Grbl_Esp32/files/9688111/mpcnc_v1p2.txt)

MitchBradley, 2022-09-30:
Spell it the way I said

graham2550, 2022-10-01:
Thanks for the help, appreciated.
I missed the underscore!
I'll try again.
regards
Graham Payne
0413 194 244