bdring/Grbl_Esp32 #1292: Translate Machine.h for MY_Laser_translated

graham2550, 2022-10-01:
This issue will automatically convert a Grbl_Esp32 machine definition file to FluidNC YAML format.

Change YOUR NAME in the title to some string that identifies your machine configuration.

IMPORTANT: Change only the YOUR NAME part of the title, not the entire title.  If you change the entire title, GitHub will not run the converter program.

Attach your machine definition file here - but first change the .h extension to .txt because GitHub will not allow you to attach a file with a .h extension.  Submit the issue, and after a few minutes, a new comment will appear with the FluidNC YAML file.

IMPORTANT: Attach the file - do not paste the file text into the issue.  If you paste, the converter will not work.
[mpcnc_v1p2.txt](https://github.com/bdring/Grbl_Esp32/files/9689315/mpcnc_v1p2.txt)

graham2550, 2022-10-01:
I have a couple of queries.
1. I cannot see machine settings in any of my control software, Lightburn or UGS.
2. Lightburn sits waiting for connection even though I have USB set as the connection type.
3. can you advise please

MitchBradley, 2022-10-01:
If you are asking about FluidNC, please file an issue on the FluidNC repo and provide all of the information that it asks for.