bdring/Grbl_Esp32 #1299: Allow Control Signals via I2S

gxdeange, 2022-10-07:
Hi Team,

Is there a way to send control signals (such as SD Card CS, Spindle Enable and Spindle Direction via I2S. I have managed to get Flood and Mist working.

I am building a custom board and need to conserve as many GPIO's (for Inputs) as possible.

Cheers
George

MitchBradley, 2022-10-07:
Spindle controls should word via I2S.  Maybe not SD Card CS, because the SD Card driver is buried deep in the ESP-IDF framework and assumes that the SD CS pin is a GPIO.  With a lot of effort, we were able to make the Trinamic CS pins works from I2S, but I wouldn't want to try to do that level of trickery for the SD driver.

gxdeange, 2022-10-07:
Thanks for your feedback Mitch.

I did try the Spin Enable and Direction but when I tried to compile it was
throwing an error that it could not convert the pin assignment.

Do I need to define anything within the .h file prior to compiling?

Cheers
George

MitchBradley, 2022-10-07:
Oh, sorry, I didn't notice that you are still on Grbl_ESP32.  It has been so long since we switched our development over to FluidNC that I do not remember what Grbl_ESP32 does in that case.  Actually, maybe I do sort of remember - the full generalization of pins was one of the main drivers of the switch to FluidNC, so I am not surprised that you have problems with Grbl_ESP32 and I2SO pins in odd places.

gxdeange, 2022-10-07:
Ok. No problem.

Thanks for taking your time to respond.

Cheers and have a great day
George

MitchBradley, 2022-10-07:
If you stick with Grbl_ESP32 you will be further and further behind the power curve.  In addition to many new features, there are bugs that we fixed in FluidNC that we do not intend to backport to Grbl_ESP32.

closed 2022-10-07 by MitchBradley