bdring/Grbl_Esp32 #1312: File index.html.gz is missing, please upload it

onisuk, 2022-11-04:
HI I've got an Atomstack X7 and it looks like they use this firmware, the web interface config is missing on startup as expected, I have uploaded the file from here and this got me the working interface that can get me into config and move the laser head about etc...

but as this seems aimed at cnc machines is there a version of this index config file for laser machines? I have seen in their instructions that they have a version with laser power and speed controls, i have emailed "tech support" and they have never heard of this file...

[image]

bdring, 2022-11-04:
We have no affiliation with AtomStack. If they are using our firmware or a modification of the firmware, they must publish the source code or they are in violation of the license.

Grbl_ESP32 has been upgraded to FluidNC, so we no longer maintain Grbl_ESP32. The file you are looking for is in this folder, but I have no idea if it will work with your setup.

https://github.com/bdring/Grbl_Esp32/tree/main/Grbl_Esp32/data

We recommend upgrading to FluidNC.

https://github.com/bdring/FluidNC

onisuk, 2022-11-04:
Well, they are definitely using it as their controller took me to this repo.

I suggest this is something you take up with them, a classic Chinese move really...

support@atomstack.com

MitchBradley, 2022-11-04:
There is no way to "win" by getting into a license dispute with a company in a country that does not work with the rest of the world to protect IP.

quintesse, 2022-11-29:
I'm in the same situation with a laser cutter from the same company.
When you say:

> We recommend upgrading to FluidNC.

Do you have any pointers on how to go about that?
Because without the source code we'd somehow have to reverse-engineer the config file needed to build the firmware, right?