bdring/Grbl_Esp32 #1316: Translate Machine.h for My_Laser

WallyinNC, 2022-11-09:
This issue will automatically convert a Grbl_Esp32 machine definition file to FluidNC YAML format.

Change YOUR NAME in the title to some string that identifies your machine configuration.

IMPORTANT: Change only the YOUR NAME part of the title, not the entire title.  If you change the entire title, GitHub will not run the converter program.

Attach your machine definition file here - but first change the .h extension to .txt because GitHub will not allow you to attach a file with a .h extension.  Submit the issue, and after a few minutes, a new comment will appear with the FluidNC YAML file.

IMPORTANT: Attach the file - do not paste the file text into the issue.  If you paste, the converter will not work.
[GRBL_Machine.txt](https://github.com/bdring/Grbl_Esp32/files/9973271/GRBL_Machine.txt)

MitchBradley, 2022-11-09:
This time you pasted the grbl settings, not the .h file.  The .h file is the one that you had to edit or select before compiling Grbl_Esp32.  Its original filename ends in ".h" .

WallyinNC, 2022-11-09:
Thanks

I actually don’t have that, I wasn’t able to install Grbl_ESP32, I keep getting errors in compilation.

Trying to work with FluidNC now. But creating the yaml file seems to be quite challenging. 

 

Walter Bonomo

***@***.***

+1 847 420 8647

MitchBradley, 2022-11-09:
There are lots of example YAML files.  Pick one that uses the board that you have, with a similar number of axes and the same driver type.  Ask questions on discord