bdring/Grbl_Esp32 #1320: Missing FreeRTOS.h library

ItsOkeyPro, 2022-11-16:
Hello!
When I am trying to compile the grbl for esp32, it shows me this error:
Alternatives for FreeRTOS.h: []
ResolveLibrary(FreeRTOS.h)
  -> candidates: []
C:\Users\ewleg\Downloads\Grbl_Esp32-main\Grbl_Esp32-main\Grbl_Esp32\src\I2SOut.cpp:54:10: fatal error: FreeRTOS.h: No such file or directory
 #include <FreeRTOS.h>
          ^~~~~~~~~~~~
compilation terminated.

I tried to install FreeRTOS library through library manager, but it is called Arduino_FreeRTOS.h, so it doesnt find it. When I try to rename it to FreeRTOS.h it shows error of avr/io.h missing.
Thanks!

MitchBradley, 2022-11-16:
Grbl_Esp32 is no longer supported.  The developers are focused entirely on [FluidNC](https://github.com/bdring/FluidNC), an improved version of Grbl_Esp32 that users can use without compiling, since it is configured with a text file.

synman, 2022-11-26:
what does this mean, lol?   Do we need to go back to a prior release or is there a way to jimmy jam freertos to work?

MitchBradley, 2022-11-26:
Updates to the Arduino Framework and the ESP-IDK and gcc toolchain on which it depends often break compiles.  In FluidNC land, it took us a long time to stabilize things after the recent major framework update.

There are FluidNC configurations for XPro V5.

synman, 2022-11-27:
I've been playing with them since last night.  It is quite a departure from what I'm used to.

Very quickly found myself missing $$

I had to go with an older version too as it appears the tmc stepper drivers the xPro uses are broke.

A few other oddities as well.   I can Invert my x motor but when I try to invert the pins for my two Y motors, things crash.

I could always swap my wires but it feels like a step backwards in it's current state, unfortunately.

MitchBradley, 2022-11-27:
If you file an issue on the FluidNC github with the requested information, we will address it.

synman, 2022-11-27:
Appreciate it.  There's already an open issue for the steppers.  It looks like there are some conflicts between the tmc driver and SD card access .

Perhaps I'll suggest I don't care about SD card support, but it appears the problem is a bit more complicated than that.

DmiwX, 2025-05-28:
Здравствуйте! Я использую плату без названия, только на чипе указано ESP-32, Arduino Ide при запуске GetChipID определяет её как "_Микросхема ESP32-D0WD-V3 (ревизия v3.1)_"  Но мне необходима установить GRBL Esp32. Использую Arduino ID м.2,3,6 и все последние библиотеки. С Github скачиваю Grbl_Esp32, распаковываю в C:\Users\user\Documents\Arduino кидаю 2 библиотеки с архива в C:\Users\user\Documents\Arduino\libraries, устанавливаю TMCStepper, компиляция с ошибкой. Подскажите пожалуйста куда копать? (детализированный вывод):

FQBN: esp32:esp32:esp32
использование платы 'esp32' из платформы в каталоге: C:\Users\DmX_X\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0
Использование ядра 'esp32' из платформы в каталоге: C:\Users\DmX_X\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0

cmd /c if exist "C:\\Users\\DmX_X\\Documents\\Arduino2\\Grbl_Esp32\\partitions.csv" COPY /y "C:\\Users\\DmX_X\\Documents\\Arduino2\\Grbl_Esp32\\partitions.csv" "C:\\Users\\DmX_X\\AppData\\Local\\arduino\\sketches\\E51BB94EF044F08AE2862F47002145AB\\partitions.csv"
cmd /c if not exist "C:\\Users\\DmX_X\\AppData\\Local\\arduino\\sketches\\E51BB94EF044F08AE2862F47002145AB\\partitions.csv" if exist "C:\\Users\\DmX_X\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.2.0\\variants\\esp32\\partitions.csv" COPY "C:\\Users\\DmX_X\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.2.0\\variants\\esp32\\partitions.csv" "C:\\Users\\DmX_X\\AppData\\Local\\arduino\\sketches\\E51BB94EF044F08AE2862F47002145AB\\partitions.csv"
cmd /c if not exist "C:\\Users\\DmX_X\\AppData\\Local\\arduino\\sketches\\E51BB94EF044F08AE2862F47002145AB\\partitions.csv" COPY "C:\\Users\\DmX_X\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.2.0\\tools\\partitions\\default.csv" "C:\\Users\\DmX_X\\AppData\\Local\\arduino\\sketches\\E51BB94EF044F08AE2862F47002145AB\\partitions.csv"
cmd /c IF EXIST "C:\\Users\\DmX_X\\Documents\\Arduino2\\Grbl_Esp32\\bootloader.bin" ( COPY /y "C:\\Users\\DmX_X\\Documents\\Arduino2\\Grbl_Esp32\\bootloader.bin" "C:\\Users\\DmX_X\\AppData\\Local\\arduino\\sketches\\E51BB94EF044F08AE2862F47002145AB\\Grbl_Esp32.ino.bootloader.bin" ) ELSE ( IF EXIST "C:\\Users\\DmX_X\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.2.0\\variants\\esp32\\bootloader.bin" ( COPY "C:\\Users\\DmX_X\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.2.0\\variants\\esp32\\bootloader.bin" "C:\\Users\\DmX_X\\AppData\\Local\\arduino\\sketches\\E51BB94EF044F08AE2862F47002145AB\\Grbl_Esp32.ino.bootloader.bin" ) ELSE ( "C:\\Users\\DmX_X\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esptool_py\\4.9.dev3\\esptool.exe" --chip esp32 elf2image --flash_mode dio --flash_freq 80m --flash_size 4MB -o "C:\\Users\\DmX_X\\AppData\\Local\\arduino\\sketches\\E51BB94EF044F08AE2862F47002145AB\\Grbl_Esp32.ino.bootloader.bin" "C:\\Users\\DmX_X\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.4-2f7dcd86-v1\\esp32\\bin\\bootloader_qio_80m.elf" ) )
esptool.py v4.8.1
Creating esp32 image...
Merged 2 ELF sections
Successfully created esp32 image.
cmd /c if exist "C:\\Users\\DmX_X\\Documents\\Arduino2\\Grbl_Esp32\\build_opt.h" COPY /y "C:\\Users\\DmX_X\\Documents\\Arduino2\\Grbl_Esp32\\build_opt.h" "C:\\Users\\DmX_X\\AppData\\Local\\arduino\\sketches\\E51BB94EF044F08AE2862F47002145AB\\build_opt.h"
cmd /c if not exist "C:\\Users\\DmX_X\\AppData\\Local\\arduino\\sketches\\E51BB94EF044F08AE2862F47002145AB\\build_opt.h" type nul > "C:\\Users\\DmX_X\\AppData\\Local\\arduino\\sketches\\E51BB94EF044F08AE2862F47002145AB\\build_opt.h"
cmd /c type nul > "C:\\Users\\DmX_X\\AppData\\Local\\arduino\\sketches\\E51BB94EF044F08AE2862F47002145AB/file_opts"
cmd /c COPY /y "C:\\Users\\DmX_X\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.4-2f7dcd86-v1\\esp32\\sdkconfig" "C:\\Users\\DmX_X\\AppData\\Local\\arduino\\sketches\\E51BB94EF044F08AE2862F47002145AB\\sdkconfig"
�����஢��� 䠩���:         1.
Обнаружение используемых библиотек...
C:\Users\DmX_X\AppData\Local\Arduino15\packages\esp32\tools\esp-x32\2411/bin/xtensa-esp32-elf-g++ -c @C:\Users\DmX_X\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.4-2f7dcd86-v1\esp32/flags/cpp_flags -w -Os -Werror=return-type -w -x c++ -E -CC -DF_CPU=240000000L -DARDUINO=10607 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32 -DARDUINO_BOARD="ESP32_DEV" -DARDUINO_VARIANT="esp32" -DARDUINO_PARTITION_default -DARDUINO_HOST_OS="windows" -DARDUINO_FQBN="esp32:esp32:esp32:UploadSpeed=921600,CPUFreq=240,FlashFreq=80,FlashMode=qio,FlashSize=4M,PartitionScheme=default,DebugLevel=none,PSRAM=disabled,LoopCore=1,EventsCore=1,EraseFlash=none,JTAGAdapter=default,ZigbeeMode=default" -DESP32=ESP32 -DCORE_DEBUG_LEVEL=0 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1 -DARDUINO_USB_CDC_ON_BOOT=0 @C:\Users\DmX_X\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.4-2f7dcd86-v1\esp32/flags/defines -IC:\Users\DmX_X\Documents\Arduino2\Grbl_Esp32 -iprefix C:\Users\DmX_X\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.4-2f7dcd86-v1\esp32/include/ @C:\Users\DmX_X\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.4-2f7dcd86-v1\esp32/flags/includes -IC:\Users\DmX_X\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.4-2f7dcd86-v1\esp32/qio_qspi/include -IC:\Users\DmX_X\AppData\Local\Arduino15\pack [... cut ...] 1BB94EF044F08AE2862F47002145AB/file_opts C:\Users\DmX_X\AppData\Local\arduino\sketches\E51BB94EF044F08AE2862F47002145AB\sketch\src\I2SOut.cpp -o nul
C:\Users\DmX_X\Documents\Arduino2\Grbl_Esp32\src\I2SOut.cpp:54:10: fatal error: FreeRTOS.h: No such file or directory
   54 | #include <FreeRTOS.h>
      |          ^~~~~~~~~~~~
compilation terminated.
Альтернативы для FreeRTOS.h: []
ResolveLibrary(FreeRTOS.h)
  -> кандидаты: []
exit status 1

Compilation error: FreeRTOS.h: No such file or directory

MitchBradley, 2025-05-28:
We are no longer supporting Grbl_Esp32.  FluidNC is the supported code.