bdring/Grbl_Esp32 #1324: Translate Machine.h for CNC_xPRO_V5_XYYZ_NC

synman, 2022-11-26:
This issue will automatically convert a Grbl_Esp32 machine definition file to FluidNC YAML format.

Change YOUR NAME in the title to some string that identifies your machine configuration.

IMPORTANT: Change only the YOUR NAME part of the title, not the entire title.  If you change the entire title, GitHub will not run the converter program.

Attach your machine definition file here - but first change the .h extension to .txt because GitHub will not allow you to attach a file with a .h extension.  Submit the issue, and after a few minutes, a new comment will appear with the FluidNC YAML file.

IMPORTANT: Attach the file - do not paste the file text into the issue.  If you paste, the converter will not work.
[CNC_xPRO_V5_XYYZ_NC.txt](https://github.com/bdring/Grbl_Esp32/files/10095918/CNC_xPRO_V5_XYYZ_NC.txt)

synman, 2022-11-27:
This did not go well.  I had to adapt the same config.yaml for the xPro provided in the wiki and had to fall back a couple versions in fluidnc to get it to recognize my stepper drivers.

MitchBradley, 2022-11-27:
Report such issues on the FluidNC github.