bdring/Grbl_Esp32 #1339: Translate Machine.h for Open Builds BlackBox X32

synman, 2022-12-13:
This issue will automatically convert a Grbl_Esp32 machine definition file to FluidNC YAML format.

Change YOUR NAME in the title to some string that identifies your machine configuration.

IMPORTANT: Change only the YOUR NAME part of the title, not the entire title.  If you change the entire title, GitHub will not run the converter program.

Attach your machine definition file here - but first change the .h extension to .txt because GitHub will not allow you to attach a file with a .h extension.  Submit the issue, and after a few minutes, a new comment will appear with the FluidNC YAML file.

IMPORTANT: Attach the file - do not paste the file text into the issue.  If you paste, the converter will not work.

[OpenBuilds_BlackBox_X32-XYYZ-DOOR.txt](https://github.com/bdring/Grbl_Esp32/files/10214932/OpenBuilds_BlackBox_X32-XYYZ-DOOR.txt)

MitchBradley, 2022-12-16:
I think I fixed the converter so if you want to you can try again (open a new issue).

synman, 2022-12-16:
failed here:

```
Grbl_Esp32/src/System.cpp: In function ‘void system_ini()’:
[91](https://github.com/bdring/Grbl_Esp32/actions/runs/3717109441/jobs/6304152985#step:8:92)
Grbl_Esp32/src/System.cpp:103:70: error: no matching function for call to ‘SPIClass::begin(int, int, int, int)’
[92](https://github.com/bdring/Grbl_Esp32/actions/runs/3717109441/jobs/6304152985#step:8:93)
  103 |     SPI.begin(GRBL_SPI_SCK, GRBL_SPI_MISO, GRBL_SPI_MOSI, GRBL_SPI_SS);
[93](https://github.com/bdring/Grbl_Esp32/actions/runs/3717109441/jobs/6304152985#step:8:94)
      |                                                                      ^
[94](https://github.com/bdring/Grbl_Esp32/actions/runs/3717109441/jobs/6304152985#step:8:95)
In file included from libraries/NativeStubs/SD.h:2,
[95](https://github.com/bdring/Grbl_Esp32/actions/runs/3717109441/jobs/6304152985#step:8:96)
                 from Grbl_Esp32/src/SDCard.h:20,
[96](https://github.com/bdring/Grbl_Esp32/actions/runs/3717109441/jobs/6304152985#step:8:97)
                 from Grbl_Esp32/src/Grbl.h:80,
[97](https://github.com/bdring/Grbl_Esp32/actions/runs/3717109441/jobs/6304152985#step:8:98)
                 from Grbl_Esp32/src/System.cpp:21:
[98](https://github.com/bdring/Grbl_Esp32/actions/runs/3717109441/jobs/6304152985#step:8:99)
.pio/libdeps/native/EpoxyDuino/cores/epoxy/SPI.h:80:15: note: candidate: ‘static void SPIClass::begin()’
[99](https://github.com/bdring/Grbl_Esp32/actions/runs/3717109441/jobs/6304152985#step:8:100)
   80 |   static void begin() { }
[100](https://github.com/bdring/Grbl_Esp32/actions/runs/3717109441/jobs/6304152985#step:8:101)
      |               ^~~~~
[101](https://github.com/bdring/Grbl_Esp32/actions/runs/3717109441/jobs/6304152985#step:8:102)
.pio/libdeps/native/EpoxyDuino/cores/epoxy/SPI.h:80:15: note:   candidate expects 0 arguments, 4 provided
[102](https://github.com/bdring/Grbl_Esp32/actions/runs/3717109441/jobs/6304152985#step:8:103)
*** [.pio/build/native/src/src/System.o] Error 1
```

MitchBradley, 2022-12-16:
Comment out `#define GRBL_SPI_SS 5`.  It really isn't correct to apply that globally to the SPI bus because the chip select is different for each device that uses SPI.  GPIO 5 will be applied later when the SD card driver is attached.

synman, 2022-12-16:
I think I can delete all of these ... I put them in based on a convo with OB.   I think they default to the same values anyway...

```
#define GRBL_SPI_FREQ 4000000
#define GRBL_SPI_SS 5
#define GRBL_SPI_MOSI 23
#define GRBL_SPI_MISO 19
#define GRBL_SPI_SCK 18
```

synman, 2022-12-16:
For reference, here's the config.yaml I hand coded the other night.  It became the basis for my current config (which is rock solid :))

```
board: OpenBuilds BlackBox X32
name: BlackBox_X32_XYYZ-DOOR
meta: Handcrafted by Shell M Shrader
stepping:
  engine: RMT
  idle_ms: 255
  pulse_us: 2
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.17:low
  x:
    steps_per_mm: 200.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      feed_mm_per_min: 100.000
      seek_mm_per_min: 500.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: gpio.12
        direction_pin: gpio.14

  y:
    steps_per_mm: 200.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      feed_mm_per_min: 100.000
      seek_mm_per_min: 500.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: gpio.27
        direction_pin: gpio.26

    motor1:
      standard_stepper:
        step_pin: gpio.33
        direction_pin: gpio.32

  z:
    steps_per_mm: 200.000
    max_rate_mm_per_min: 300.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 50.000
    soft_limits: false
    homing:
      cycle: 1
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.39
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: gpio.15
        direction_pin: gpio.2

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

probe:
  pin: gpio.22:low
  check_mode_start: true

control:
  safety_door_pin: gpio.16:pu

macros:
  startup_line0:
  startup_line1:
  macro0:
  macro1:
  macro2:
  macro3:

start:
  must_home: false

coolant:
  mist_pin: gpio.21
  delay_ms: 0

PWM:
  pwm_hz: 5000
  output_pin: gpio.25
  enable_pin: gpio.13
  direction_pin: gpio.4
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 30000
  spindown_ms: 10000
  tool_num: 0
  speed_map: 0=0.000% 1000=100.000%
```

synman, 2022-12-16:
ran fine without GRBL_SPI_SS ... gonna close this.  thanks @MitchBradley !

closed 2022-12-16 by synman

synman, 2022-12-16:
I'm not sure if it matter, but I did notice the generator configured my steppers as "stepstick" when I went with "standard_stepper".

I'm guessing this would be similar to the differences between a PWM and 10V spindle (not many).