bdring/Grbl_Esp32 #1379: Translate Machine.h for hello

Ammar56788, 2023-02-01:
[Machine.txt](https://github.com/bdring/Grbl_Esp32/files/10561667/Machine.txt)
This issue will automatically convert a Grbl_Esp32 machine definition file to FluidNC YAML format.

Change YOUR NAME in the title to some string that identifies your machine configuration.

IMPORTANT: Change only the YOUR NAME part of the title, not the entire title.  If you change the entire title, GitHub will not run the converter program.

Attach your machine definition file here - but first change the .h extension to .txt because GitHub will not allow you to attach a file with a .h extension.  Submit the issue, and after a few minutes, a new comment will appear with the FluidNC YAML file.

IMPORTANT: Attach the file - do not paste the file text into the issue.  If you paste, the converter will not work.

closed 2023-02-01 by Ammar56788

kami0011, 2023-06-30:
Hi.
I am moving from grbl esp32 to fluid . Please convert my file to yaml.
[convertp.txt](https://github.com/bdring/Grbl_Esp32/files/11917081/convertp.txt)

my hardware is using esp32 dev 30 pin .
When i change yaml to configure according to my pins it wont work.
I think fluidnc is configured using 48 pin. Please help me in this regard.
Thanks a lot

MitchBradley, 2023-06-30:
@kami0011 You must file a new issue and attach your file to that, instead of replying to an existing issue.

[image]