bdring/Grbl_Esp32 #1380: Translate Machine.h for just whom

Ammar56788, 2023-02-01:
[atari_1020.txt](https://github.com/bdring/Grbl_Esp32/files/10561687/atari_1020.txt)
This issue will automatically convert a Grbl_Esp32 machine definition file to FluidNC YAML format.

Change YOUR NAME in the title to some string that identifies your machine configuration.

IMPORTANT: Change only the YOUR NAME part of the title, not the entire title.  If you change the entire title, GitHub will not run the converter program.

Attach your machine definition file here - but first change the .h extension to .txt because GitHub will not allow you to attach a file with a .h extension.  Submit the issue, and after a few minutes, a new comment will appear with the FluidNC YAML file.

IMPORTANT: Attach the file - do not paste the file text into the issue.  If you paste, the converter will not work.

MitchBradley, 2023-02-02:
One possible problem is that atari_1020.txt specifies Unipolar steppers.  We stopped supporting unipolar because it is so rarely used and there are much better choices now.  The effort of maintaining the unipolar code for so few users was not worth it.

MitchBradley, 2023-02-02:
The Grbl_Esp32 atari_1020 configuration used custom code that we did not port to FluidNC.  For that machine you will need to continue using Grbl_Esp32.

Ammar56788, 2023-02-02:
I need simplest unipolar stepper with just xy axis 28byj-48 and pen servo sg90 and a limit switch for each xy axis (control by PC microUSB or webUI). Without any other thing such as color sw, pen sw, paper sw, solenoid dir, microSD and etc. 
I don't know what i must to do and how edit Atari_1020.h and .cs (custom folder) and make new config as .h or .bin etc.
and .bin file that uplod to main spiffs by OTA include what?