bdring/Grbl_Esp32 #1386: Edit Atari_1020.h to simplest case and write new config

Ammar56788, 2023-02-02:
I need simplest unipolar-stepper pen plotter with just xy axis 28byj-48 and pen servo sg90 and a limit switch for each xy axis (control by PC microUSB or webUI).
Without any other thing such as color sw, pen sw, paper sw, solenoid dir, microSD and etc.
I don't know what i must to do and how edit Atari_1020.h and .cs (custom folder) and make new config as .h or .bin etc.
and .bin file that uplod to main spiffs by OTA include what?

_Originally posted by @Ammar56788 in https://github.com/bdring/Grbl_Esp32/issues/1380#issuecomment-1414345735_

MitchBradley, 2023-02-02:
We are not supporting Grbl_Esp32 any more.  Good luck.

closed 2023-02-02 by MitchBradley

Ammar56788, 2023-02-02:
I need it for fluidnc if support unipolar.