bdring/Grbl_Esp32 #1410: Translate Machine.h for MJ666

MJ666, 2023-02-27:
[i2s_out_xyz_laserbox_esp32.TXT](https://github.com/bdring/Grbl_Esp32/files/10845036/i2s_out_xyz_laserbox_esp32.TXT)

MitchBradley, 2023-02-28:
The converter cannot handle configurations that include LCD support.  LCD support is not present in the mainline Grbl_Esp32 code.  It was added by a vendor's private code.  I think that the board you are using might be a clone of MKS DLC32, so you could start with one of the DLC32 configurations in FluidNC/example_configs and modify it to suit your machine.

MJ666, 2023-03-01:
> The converter cannot handle configurations that include LCD support. LCD support is not present in the mainline Grbl_Esp32 code. It was added by a vendor's private code. I think that the board you [...]

Many thanks. Already created the file manualy.