bdring/Grbl_Esp32 #1488: CNCxProV5

Mercydoc, 2023-06-13:
[WorkBee V3 1000x1500 Standard Torque.txt](https://github.com/bdring/Grbl_Esp32/files/11733758/WorkBee.V3.1000x1500.Standard.Torque.txt)
This issue will automatically convert a Grbl_Esp32 machine definition file to FluidNC YAML format.

Change YOUR NAME in the title to some string that identifies your machine configuration.

IMPORTANT: Change only the YOUR NAME part of the title, not the entire title.  If you change the entire title, GitHub will not run the converter program.

Attach your machine definition file here - but first change the .h extension to .txt because GitHub will not allow you to attach a file with a .h extension.  Submit the issue, and after a few minutes, a new comment will appear with the FluidNC YAML file.

IMPORTANT: Attach the file - do not paste the file text into the issue.  If you paste, the converter will not work.

MitchBradley, 2023-06-13:
You must post the machine.h file, not the $$ settings file.

MitchBradley, 2023-06-13:
Please also note that the XProV5 manufacturer has information about FluidNC, including a sample config file, at https://github.com/Spark-Concepts/xPro-V5/tree/main/FluidNC