bdring/Grbl_Esp32 #1497: Translate Machine.h for YOUR NAME

kami0011, 2023-06-30:
This issue will automatically convert a Grbl_Esp32 machine definition file to FluidNC YAML format.

Change YOUR NAME in the title to some string that identifies your machine configuration.

IMPORTANT: Change only the YOUR NAME part of the title, not the entire title.  If you change the entire title, GitHub will not run the converter program.

Attach your machine definition file here - but first change the .h extension to .txt because GitHub will not allow you to attach a file with a .h extension.  Submit the issue, and after a few minutes, a new comment will appear with the FluidNC YAML file.

IMPORTANT: Attach the file - do not paste the file text into the issue.  If you paste, the converter will not work.

hi
Thank you so much for your prompt reply.
Actually this project i am using on a multiaxis pick and place machine.
am not using spindle or extruder in this code.
on inputs control i am using 4 control inputs and i use control 1 for home all axis , 2 for clear alarm, 3 for emergency stop , and 4 for run .
also i using 3 outputs to run through M62 P0, P1,P2,P3
This is actually i am running on grbl ESP32
Please see attached file for my pcb design. I am using 30 pin ESP32 Wroom.
Is this fluidNC running on ESP32 Dev Controller V4 which is 48 pin .
I cant change my hardware. So please suggest what to do .
I have old GRBL Esp32 code but having so many problems compiling that.
Please see attached PCB
[PCB_PCB4A_2023-06-30 (1).pdf](https://github.com/bdring/Grbl_Esp32/files/11918647/PCB_PCB4A_2023-06-30.1.pdf)
[PCB_PCB4A_2023-06-30.pdf](https://github.com/bdring/Grbl_Esp32/files/11918666/PCB_PCB4A_2023-06-30.pdf)
[esp32 30pin.pdf](https://github.com/bdring/Grbl_Esp32/files/11918672/esp32.30pin.pdf)

Really appreciate for all your help.
Thanks

MitchBradley, 2023-06-30:
File an Problem issue on the FluidNC repo. Provide all the information it asks for, attach your photos and describe the problems you are having in great detail.

kami0011, 2023-06-30:
Hi,
I have just explained my problem and also attached pictures of hardware aswell.
Can you please help in this regard
I am using 30 pin ESP32 . Will it work on FluidNC. If yes then what do i have to do to make it work. I have YAML file already configured