bdring/Grbl_Esp32 #1516: Translate Machine.h for Stavros' TimSav

skorokithakis, 2023-08-22:
This issue will automatically convert a Grbl_Esp32 machine definition file to FluidNC YAML format.

Change YOUR NAME in the title to some string that identifies your machine configuration.

IMPORTANT: Change only the YOUR NAME part of the title, not the entire title.  If you change the entire title, GitHub will not run the converter program.

Attach your machine definition file here - but first change the .h extension to .txt because GitHub will not allow you to attach a file with a .h extension.  Submit the issue, and after a few minutes, a new comment will appear with the FluidNC YAML file.

IMPORTANT: Attach the file - do not paste the file text into the issue.  If you paste, the converter will not work.
[timsav.txt](https://github.com/bdring/Grbl_Esp32/files/12409623/timsav.txt)

MitchBradley, 2023-08-24:
I made the requested change.  Please test it.

skorokithakis, 2023-08-24:
It works, thank you.