bdring/Grbl_Esp32 #1520: Features: Ethernet communication. Asking for the help to understand the difficulty of the task and ways to have it solved.

romulie, 2023-09-09:
Hello!

The main goal is: to have the possibility to upload job files using Ethernet (not WiFi because of interference issues and long distances) and start them on grbl_esp32.

Yes, I have read about the plans to add the Ethernet connectivity to grbl_esp32 in some distant future. I tried it myself but with no success yet.

I’m not a very experienced programmer and I don’t fully understand the TCP\IP stack etc and internal guts of grbl_esp32 code. I have made some tries to have the feature working:
1. I connected SPI Ethernet shield based on W5500 to esp32. I used 10cm short wires
2. Used the library <Ethernet2> suitable for the W5500
3. Configured Ethernet using Ethernet.init(W5500_CS_PIN) and Ethernet.begin(mac, IP) method BEFORE and AFTER the grbl_init() function in setup();
4. Made some manipulations with the library <Ethernet2> since there are some problems in it (example sketches for the Ethernet server with W5500 just didn’t work)
5. I noticed problems to start Ethernet when the SD-card reader is on the same SPI despite the separate chip-select pins used for card-reader and Ethernet shield.
6. I Read about problems with W5500 and SPIFFS on ESP32 and some other stuff…
7. I have a ENC28J60 Ethernet board and it worked on my other project with ESP32 (not grbl) just fine but it requires newer ESP32-board version in ArduinoIDE (and grbl_esp32 requires the OLDER one…).

So I’m running out of ideas and need some tips.

I would be grateful for:
1. The general outline of the problems to have the Ethernet connectivity added to the existing code:
- Is this a hardware or a software problem?
- Is this an architectural problem to insert the Ethernet functionality into existing code without breaking it apart?
2. Because I don’t fully understand the logic and sequencing of the events in grbl concerning network connections, clients, configurations etc, could someone describe me the general flow of events in grbl_esp32.
3. Steps to be taken to solve the task.
4. Would it be much simpler to leave only Ethernet connectivity and discard all WiFi features for my local version of grbl_esp32?

Thank you in advance guys.

BTW: I managed to add I2C OLED Display and simple joystick control to grbl_esp32 using some features of FreeRTOS and manipulations with guts of the existing code. Sure enough my implementation is not a good one and needs lots of refinements. But it somehow works and may be helpful for the people who want to have the features of this kind: https://github.com/romulie/ESP32_grbl_OLED_and_Joystick

MitchBradley, 2023-09-09:
Grbl_Esp32 is no longer maintained.  All development has moved to it successor FluidNC.

MitchBradley, 2023-09-10:
If you open a similar issue on the FluidNC repo, I will tell you the considerations.

romulie, 2023-09-11:
Thank you! I'll do.