bdring/Grbl_Esp32 #1528: Licensing question

G-EDM, 2023-10-05:
Hi,  

I modified the codebase a little and put some custom code next it as a different lib. The codebase I created is not meant to be published under GPL. Mainly to prevent business men from just ripping it. The code is only published on Github for personal use and research and will never be shipped or sold in any products that may emerge or provided as a ready to use compiled  application.

I just would like to have a confirmation that you are ok with this. This is the critical section of the project:
https://github.com/G-EDM/G-EDM/tree/main/Software

If this is not ok for you I will remove the grbl code.

Thanks.

bdring, 2023-10-05:
You are free to use our code however you want. The only thing you cannot do
is change the license on any of our files. They must retain the original
license even if you add your own code. If you reference outside code, that
code can have your license.

We have businesses "rip" our code all the time. It generally results in a
positive outcome. I encourage you to join the free and open source
community.

G-EDM, 2023-10-05:
Hi. Thanks. The changes that where made to the core grbl files are of course accepting the license. It is basically just a small set of files like the UI and the new planner that are excluded. At some point all files will be opensource. But for now I want to have at least a tiny bit of control.

Highly appreciate your work and as claimed on the front readme: If donations are coming in, you will get a fair share of it.