bdring/Grbl_Esp32 #1531: Translate Machine.h for chrisa10proatomstacknolcdgpio15

arraylabs, 2023-10-06:
This issue will automatically convert a Grbl_Esp32 machine definition file to FluidNC YAML format.

Change YOUR NAME in the title to some string that identifies your machine configuration.

IMPORTANT: Change only the YOUR NAME part of the title, not the entire title.  If you change the entire title, GitHub will not run the converter program.

Attach your machine definition file here - but first change the .h extension to .txt because GitHub will not allow you to attach a file with a .h extension.  Submit the issue, and after a few minutes, a new comment will appear with the FluidNC YAML file.

IMPORTANT: Attach the file - do not paste the file text into the issue.  If you paste, the converter will not work.
[i2s_out_xyz_laserbox_esp32.TXT](https://github.com/bdring/Grbl_Esp32/files/12826883/i2s_out_xyz_laserbox_esp32.TXT)

arraylabs, 2023-10-06:
If anyone has any thoughts that would be great, tried a couple times removing the LCD and SDcard GPIO 15 stuff I saw in some other posts.

MitchBradley, 2023-10-07:
The translator sometimes gets tangled up with machine defs that use uncommon features.  You will probably need to make a config manually.  There are many example config files for various machines.  The config file format is very well documented.