bdring/Grbl_Esp32 #1551: Translate Machine.h for drawbot

senthiru, 2023-11-24:
[drawbot.txt](https://github.com/bdring/Grbl_Esp32/files/13456889/drawbot.txt)
This issue will automatically convert a Grbl_Esp32 machine definition file to FluidNC YAML format.

Change YOUR NAME in the title to some string that identifies your machine configuration.

IMPORTANT: Change only the YOUR NAME part of the title, not the entire title.  If you change the entire title, GitHub will not run the converter program.

Attach your machine definition file here - but first change the .h extension to .txt because GitHub will not allow you to attach a file with a .h extension.  Submit the issue, and after a few minutes, a new comment will appear with the FluidNC YAML file.

IMPORTANT: Attach the file - do not paste the file text into the issue.  If you paste, the converter will not work.

senthiru, 2023-12-26:
[X-axis moves exactly twice as much as the Y axis](https://forum.whadda.com/t/x-axis-moves-exactly-twice-as-much-as-the-y-axis/12134)

MitchBradley, 2023-12-26:
Adjust the steps_per_mm values until the axes move that amount that the Gcode command or jog command says.  If you ask to jog 10 mm and it moves 5 mm instead, double the steps_per_mm value for that axis.  Do it for each axis

senthiru, 2023-12-27:
HI, thanks for the reply, I am running on core xy drawing machine controller without jumpers for micro step, when individual y axis setps/mm- the machine moves diagonally.

senthiru, 2023-12-27:
Thanks Brad, it worked, thank you

senthiru, 2024-01-10:
my x axis is moving 2 times as I set my  kinematics to midbot how to I get both the axis to move  equal distances