bdring/Grbl_Esp32 #1559: Translate Machine.h for YOUR NAME

alohol0071, 2023-12-21:
[dlc_cfg.txt](https://github.com/bdring/Grbl_Esp32/files/13746802/dlc_cfg.txt)
This issue will automatically convert a Grbl_Esp32 machine definition file to FluidNC YAML format.

Change YOUR NAME in the title to some string that identifies your machine configuration.

IMPORTANT: Change only the YOUR NAME part of the title, not the entire title.  If you change the entire title, GitHub will not run the converter program.

Attach your machine definition file here - but first change the .h extension to .txt because GitHub will not allow you to attach a file with a .h extension.  Submit the issue, and after a few minutes, a new comment will appear with the FluidNC YAML file.

IMPORTANT: Attach the file - do not paste the file text into the issue.  If you paste, the converter will not work.

alohol0071, 2023-12-21:
[dlc_cfg.txt](https://github.com/bdring/Grbl_Esp32/files/13746803/dlc_cfg.txt)

MitchBradley, 2023-12-22:
dlc_cfg.txt is a Grbl settings file with lines like $1=n.  The translator does not handle settings files.  It handles .h files that are used when recompiling the Grbl_ESP32 source code.