bdring/Grbl_Esp32 #1593: yes i have read the wiki

josefKameni, 2024-02-07:
Please answer the following questions:

Have you read the [wiki regarding how to compile](https://github.com/bdring/Grbl_Esp32/wiki/Compiling-the-firmware)?

What version of the Arduino IDE are you using? 

What version (commit date) of the [Arduino core for the ESP32](https://github.com/espressif/arduino-esp32) are you using? 

Are you using the master branch of Grbl_ESP32?

Have you made any edits or configuration changes (list them) to the  firmware?

Please paste the compiler error text here:

MitchBradley, 2024-02-07:
Your problem is probably caused by a change in a later version of the compiler.  When we stopped working on Grbl_ESP32, the C++ compiler was a generation or two older.  We don't support Grbl_ESP32 any more; all of our current work is on FluidNC.

closed 2024-02-07 by MitchBradley

josefKameni, 2024-02-07:
> Your problem is probably caused by a change in a later version of the compiler. When we stopped working on Grbl_ESP32, the C++ compiler was a generation or two older. We don't support Grbl_ESP32 any more; all of our current work is on FluidNC.

thank you for your reply. i also have older version of arduino ide so that might work?

MitchBradley, 2024-02-07:
It is up to you.  As I said, we are not supporting Grbl_ESP32 any more.

josefKameni, 2024-02-07:
> 

thank you so much on older arduino ide i have just this error but all the oders are gone:
In file included from C:\Users\Josef\Desktop\cnc_frezka\Grbl_Esp32-main\Grbl_Esp32-main\Grbl_Esp32\src\Grbl.h:75,
                 from C:\Users\Josef\Desktop\cnc_frezka\Grbl_Esp32-main\Grbl_Esp32-main\Grbl_Esp32\Grbl_Esp32.ino:22:
C:\Users\Josef\Desktop\cnc_frezka\Grbl_Esp32-main\Grbl_Esp32-main\Grbl_Esp32\src\WebUI\BTConfig.h:23:6: error: #error Bluetooth is not enabled! Please run `make menuconfig` to and enable it
 #    error Bluetooth is not enabled! Please run `make menuconfig` to and enable it
      ^~~~~
exit status 1
Error compiling for board ESP32S2 Dev Module.

josefKameni, 2024-03-17:
Hi. I tried older version still the same error so i tried diferent versions of esp32 packages and i got diferent errors. Now i am trying version 1,0,2 and i got this error: 
In file included from C:\Users\Josef\Desktop\cnc_frezka\Grbl_Esp32-main\Grbl_Esp32-main\Grbl_Esp32\src\Grbl.h:54:0,
                 from C:\Users\Josef\Desktop\cnc_frezka\Grbl_Esp32-main\Grbl_Esp32-main\Grbl_Esp32\Grbl_Esp32.ino:22:
C:\Users\Josef\Desktop\cnc_frezka\Grbl_Esp32-main\Grbl_Esp32-main\Grbl_Esp32\src\Uart.h:39:19: error: 'size_t Uart::readBytes(char*, size_t)' marked 'override', but does not override
     size_t        readBytes(char* buffer, size_t length) override;
                   ^

exit status 1

Compilation error: 'size_t Uart::readBytes(char*, size_t)' marked 'override', but does not override