bdring/Grbl_Esp32 #1594: Translate Machine.h for TTC 3018S

spideyar, 2024-02-11:
This issue will automatically convert a Grbl_Esp32 machine definition file to FluidNC YAML format.

Change YOUR NAME in the title to some string that identifies your machine configuration.

IMPORTANT: Change only the YOUR NAME part of the title, not the entire title.  If you change the entire title, GitHub will not run the converter program.

Attach your machine definition file here - but first change the .h extension to .txt because GitHub will not allow you to attach a file with a .h extension.  Submit the issue, and after a few minutes, a new comment will appear with the FluidNC YAML file.

IMPORTANT: Attach the file - do not paste the file text into the issue.  If you paste, the converter will not work.
[TTC 3018S grbl 1.1h output.txt](https://github.com/bdring/Grbl_Esp32/files/14233349/TTC.3018S.grbl.1.1h.output.txt)

closed 2024-02-11 by spideyar

MitchBradley, 2024-02-11:
The translator does not convert Grbl $ settings.  It converts machine.h C language include files that determine the pinout for controller boards.

MitchBradley, 2024-02-11:
I think that the TTC 3018S comes with a MKS DLS32 controller board.  There are some example configuration files for that board in https://github.com/bdring/fluidnc-config-files/tree/main/official

spideyar, 2024-02-12:
Thanks for the feedback, @MitchBradley .  I am new at this and didn't have the machine.h file from my unit so thought I'd see if the required input was the grbl settings. 

That said, you're right, the unit does have the MKS DLS32 board (version 2.1 I think), and I initially tried using the example MKS_DLC32_21_XYZ.yaml; however, the example file for that controller board was way off for a 3018 router - which is why I tried this converter to see what happened.

In the end I got it to work after some manual trial and error.  Turns out the example file had the Y and Z inputs going the wrong way, the axes movement parameters were no where near right, and it assumes a laser is present rather than a spindle.  The machine is just a cheap little unit I bought to learn some concepts and boy did it give me a test.