bdring/Grbl_Esp32 #1603: Translate Machine.h for workbeemcr

atapi, 2024-03-05:
[workbee.txt](https://github.com/bdring/Grbl_Esp32/files/14496231/workbee.txt)
This issue will automatically convert a Grbl_Esp32 machine definition file to FluidNC YAML format.

Change YOUR NAME in the title to some string that identifies your machine configuration.

IMPORTANT: Change only the YOUR NAME part of the title, not the entire title.  If you change the entire title, GitHub will not run the converter program.

Attach your machine definition file here - but first change the .h extension to .txt because GitHub will not allow you to attach a file with a .h extension.  Submit the issue, and after a few minutes, a new comment will appear with the FluidNC YAML file.

IMPORTANT: Attach the file - do not paste the file text into the issue.  If you paste, the converter will not work.

MitchBradley, 2024-03-05:
I think the problem is that you have defined Y2_STEP_PIN but Y2_DIRECTION_PIN is commented out.

atapi, 2024-03-05:
But this config works fine on grbl_esp32, and it uses 2 motors on Y-Axis with 2 limit switches connected to single input pin. 
Works fine, apart "stress-squaring" - but my Y axis is flexible enogh for it.

Anyway, I wrote config.yaml from scratch, but as I understand it, in this hardware configuration I will lose the autosquaring function.