bdring/Grbl_Esp32 #1612: Translate Machine.h for YOUR NAME

Geneww, 2024-03-18:
#pragma once
// clang-format off
#define DEBUG // DEBUG MODE
#define MACHINE_NAME "BENDER"

#define N_AXIS 4

// enable
#define STEPPERS_DISABLE_PIN    GPIO_NUM_13
// step
#define X_STEP_PIN              GPIO_NUM_12
#define Y_STEP_PIN              GPIO_NUM_14
#define A_STEP_PIN              GPIO_NUM_27
// dir
#define X_DIRECTION_PIN         GPIO_NUM_26
#define Y_DIRECTION_PIN         GPIO_NUM_25
#define A_DIRECTION_PIN         GPIO_NUM_33
// limit
#define X_LIMIT_PIN             GPIO_NUM_16
#define Y_LIMIT_PIN             GPIO_NUM_4

// SERVO
#define Z_SERVO_PIN 		      GPIO_NUM_15
#define DEFAULT_Z_HOMING_MPOS     0.0   // $Z/Homing/MPos=5.0
#define DEFAULT_Z_STEPS_PER_MM    100.0
#define DEFAULT_Z_MAX_RATE        2000.0 // mm/min
#define DEFAULT_Z_ACCELERATION    50.0 // mm/sec^2
#define DEFAULT_Z_MAX_TRAVEL      5.5 //

//#define SPINDLE_TYPE SpindleType::NONE

// defaults
#define DEFAULT_STEP_PULSE_MICROSECONDS 3
#define DEFAULT_STEPPER_IDLE_LOCK_TIME 250 // stay on

#define DEFAULT_STEPPING_INVERT_MASK 0 // uint8_t
#define DEFAULT_DIRECTION_INVERT_MASK 0 // uint8_t
#define DEFAULT_INVERT_ST_ENABLE 0 // boolean
#define DEFAULT_INVERT_LIMIT_PINS 1 // boolean
#define DEFAULT_INVERT_PROBE_PIN 0 // boolean

#define DEFAULT_STATUS_REPORT_MASK 1

#define DEFAULT_JUNCTION_DEVIATION 0.01 // mm
#define DEFAULT_ARC_TOLERANCE 0.002 // mm
#define DEFAULT_REPORT_INCHES 0 // false

#define DEFAULT_SOFT_LIMIT_ENABLE 0 // false
#define DEFAULT_HARD_LIMIT_ENABLE 0  // false

#define DEFAULT_HOMING_CYCLE_0  bit(X_AXIS)
#define DEFAULT_HOMING_CYCLE_1  bit(Y_AXIS)

#define HOMING_FORCE_SET_ORIGIN // RESET 0,0,0

#define DEFAULT_HOMING_ENABLE 1
#define DEFAULT_HOMING_DIR_MASK 2 // move positive dir Y, negative X
#define DEFAULT_HOMING_FEED_RATE 350.0 // mm/min
#define DEFAULT_HOMING_SEEK_RATE 5000.0 // mm/min
#define DEFAULT_HOMING_DEBOUNCE_DELAY 250 // msec (0-65k)
#define DEFAULT_HOMING_PULLOFF 4.0 // mm

#define DEFAULT_SPINDLE_RPM_MAX 1000.0 // rpm
#define DEFAULT_SPINDLE_RPM_MIN 0.0 // rpm

#define DEFAULT_LASER_MODE 0 // false

#define DEFAULT_X_STEPS_PER_MM 8.889
#define DEFAULT_Y_STEPS_PER_MM 8.603
#define DEFAULT_A_STEPS_PER_MM 6.667 // This is percent in servo mode...used for calibration

#define DEFAULT_X_MAX_RATE 30000.0 // mm/min
#define DEFAULT_Y_MAX_RATE 30000.0 // mm/min
#define DEFAULT_A_MAX_RATE 8000.0 // mm/min

#define DEFAULT_X_ACCELERATION 3000.0 // mm/sec^2. 50 mm/sec^2 = 180000 mm/min^2
#define DEFAULT_Y_ACCELERATION 3000.0 // mm/sec^2
#define DEFAULT_A_ACCELERATION 1200.0 // mm/sec^2

#define DEFAULT_X_MAX_TRAVEL 900.0 // mm NOTE: Must be a positive value.
#define DEFAULT_Y_MAX_TRAVEL 900.0 // mm NOTE: Must be a positive value.
#define DEFAULT_A_MAX_TRAVEL 80.0 // This is percent in servo mode...used for calibration

MitchBradley, 2024-03-18:
As the instructions emphasize, you must attach the file to be converted, not paste it into the issue.

Geneww, 2024-03-19:
done