bdring/Grbl_Esp32 #1613: Translate Machine.h for MACHINE

Geneww, 2024-03-19:
This issue will automatically convert a Grbl_Esp32 machine definition file to FluidNC YAML format.

Change YOUR NAME in the title to some string that identifies your machine configuration.

IMPORTANT: Change only the YOUR NAME part of the title, not the entire title.  If you change the entire title, GitHub will not run the converter program.
[machine.txt](https://github.com/bdring/Grbl_Esp32/files/14645304/machine.txt)

Attach your machine definition file here - but first change the .h extension to .txt because GitHub will not allow you to attach a file with a .h extension.  Submit the issue, and after a few minutes, a new comment will appear with the FluidNC YAML file.

IMPORTANT: Attach the file - do not paste the file text into the issue.  If you paste, the converter will not work.

Geneww, 2024-03-19:
Translate complete, thank you very much！

SteveTropical, 2024-09-22:
[WorkBee 1000x1000 High Torque.txt](https://github.com/user-attachments/files/17091579/WorkBee.1000x1000.High.Torque.txt)

MitchBradley, 2024-09-22:
That is a settings file, not a machine definition file.  The translator only handles machine def files, that do things like assign pins.  For the settings, you have to edit the config file yourself.  The web installer has tools to help.