bdring/Grbl_Esp32 #1618: what is the maximum number of axes in simultaneous motion supported by ESP32_GRBL?

jessesun93, 2024-03-25:
what is the maximum number of axes in simultaneous motion supported by ESP32_GRBL? 
is there anybody have tested with XYZA in simultaneous motion? or tested XYZABC in simultaneous motion? i just find the axes coordinate data lost just a little somehow . i don't know why that happen, has that happened to anyone?
how about FluidNC with axes in simultaneous motion?

MitchBradley, 2024-03-25:
Both Grbl_Esp32 and FluidNC can run 6 axes simultaneously.  If you have found otherwise, please use FluidNC and if it is still happening there, file a FluidNC problem report giving all the information that it asks for.

closed 2024-03-25 by MitchBradley

jessesun93, 2024-03-25:
> Both Grbl_Esp32 and FluidNC can run 6 axes simultaneously. If you have found otherwise, please use FluidNC and if it is still happening there, file a FluidNC problem report giving all the information that it asks for.

Hi,MitchBradley
thank you very much for the quickly reply ,  this happened when I run a G code program repeat for 10000 times, the motion is simple just with G01 gcode like this:
G01X10.0Y10.0Z10.0A10.0B10.0C10.0F1000
G01P1.0
G0X0Y0Z0A0B0C0
G04P1.0
..........
M30
just repeat run these gcode for 10000 times, and both MPS or the WPS show the right coordinate data at the first started running, but finally the coordinate data showed the different data somehow, it's not the same data from the beginning. 
and the WEBUI coordinate data is same with my touch screen axis coordinate data, i think both coordinate data are not right, or precision lost happen only in simultaneous motion of 6 axis. (only happened in 6 axis simultaneous motion but not 3 axis simultaneous motion )
I don't know when this happened in the process running. i think i need spend more time on this and figure it out,or take some pictures more information later.

MitchBradley, 2024-03-25:
You mention different data but you did not show what it is.  We are not supporting Grbl_Esp32 anymore so we cannot help you with it.