bdring/Grbl_Esp32 #1635: Translate Machine.h for cnc

MomentumDan, 2024-05-19:
[config.txt](https://github.com/bdring/Grbl_Esp32/files/15368902/config.txt)
This issue will automatically convert a Grbl_Esp32 machine definition file to FluidNC YAML format.

Change YOUR NAME in the title to some string that identifies your machine configuration.

IMPORTANT: Change only the YOUR NAME part of the title, not the entire title.  If you change the entire title, GitHub will not run the converter program.

Attach your machine definition file here - but first change the .h extension to .txt because GitHub will not allow you to attach a file with a .h extension.  Submit the issue, and after a few minutes, a new comment will appear with the FluidNC YAML file.

IMPORTANT: Attach the file - do not paste the file text into the issue.  If you paste, the converter will not work.

MomentumDan, 2024-05-19:
[config.txt](https://github.com/bdring/Grbl_Esp32/files/15368908/config.txt)

MitchBradley, 2024-05-19:
The file that you have been pasting is a runtime settings file, not a "Machine.h" compilation configuration file.  The translator does not translate settings, it translates compilation configuration files.  The latter are used when you recompile Grbl_Esp32 for the purpose of describing things like the pinout of your hardware.